joseluis/huion-linux-drivers #1: Cursor not moving

chupinmaxime, 2018-04-10:
The click of the pen is working but the cursor does not move with the pen. Thanks!

Linux mint 18.3
Directory `/etc/X11/xorg.conf.d/` did not exist => created with the file `evdev-tablet.conf `
Output of `sudo ./huion-tablet-driver.py` with the `monitor_3` config.

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 37540
    |                          Max Y: 21120
    |                   Max pressure: 8191
    |                     Resolution: 2000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 95352
    |                          Max Y: 53645
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |      Unknown string #1: HUION_M167_170623
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name     GT-221 PRO
	Buttons               ENABLED (10)
	Scrollbar             ENABLED (1)
	Desktop notifications ENABLED
	Multi Monitor Setup   ENABLED
Setting up multiple monitors. . . Done!
	Mapped tablet area to "1920x1080 + 4480x0"

Huion Kamvas driver should now be running`
```

chupinmaxime, 2018-05-03:
No clue?

chupinmaxime, 2018-05-14:
Thank you for the answer.

1. I have 3 monitors (counting the HUION tablet)
2. I've just tried and I have the same problem:

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 37540
    |                          Max Y: 21120
    |                   Max pressure: 8191
    |                     Resolution: 2000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 95352
    |                          Max Y: 53645
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |      Unknown string #1: HUION_M167_170623
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name     GT-221 PRO
	Buttons               ENABLED (10)
	Scrollbar             ENABLED (1)
	Desktop notifications ENABLED
	Multi Monitor Setup   disabled

Huion Kamvas driver should now be running

% Main Menu %
button 0 = [menu_krita]
button 1 = [menu_gimp]
button 2 = [menu_darktable]
button 3 = 
button 4 = 
button 5 = 
button 6 = 
button 7 = 
button 8 = key Return
button 9 = key Escape
```

joseluis, 2018-05-14:
I've not tried it in Mint, but I've tried the script in several versions of Ubuntu including 18.04 and it works well. It may have something to do with the configuration of your system...

1. Just to make sure, does the pen work well in other system or OS? Have you tried it? Do you have more than one pen? Mine came with two... Just to rule out a malfunction. Alternatively, it may be good to know if you can make it work well with an Ubuntu Live CD or something like that.
1. Which kernel version do you have, and which python version do you use?

I'll think about adding a debug mode when I'm less busy at work, but for now it may be useful that after lines 275-277 in the script:
```
X = (data[8] << 16) + (data[3] << 8) + data[2]                   
Y = (data[5] << 8) + data[4]                                     
PRESS = (data[7] << 8) + data[6]
```

You could add the following line, so that it prints the detected pen movements and pressure to the console:
```
print("{} {} {}".format(X, Y, PRESS));
```
And tell me if it does output something when you use the pen over the screen.

joseluis, 2018-05-15:
Also tell me the output of the following command:

`grep evdev /var/log/Xorg*.log | tail -20`

chupinmaxime, 2018-05-15:
Here the output 
```
22:50 chupin@escudo:➜  ~ grep evdev /var/log/Xorg*.log | tail -20
/var/log/Xorg.20.log:[130669.564] (II) evdev: TypeMatrix.com USB Keyboard: Adding scrollwheel support
/var/log/Xorg.20.log:[130669.564] (**) evdev: TypeMatrix.com USB Keyboard: YAxisMapping: buttons 4 and 5
/var/log/Xorg.20.log:[130669.564] (**) evdev: TypeMatrix.com USB Keyboard: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
/var/log/Xorg.20.log:[130669.564] (**) Option "xkb_rules" "evdev"
/var/log/Xorg.20.log:[130669.564] (II) evdev: TypeMatrix.com USB Keyboard: initialized for relative axes.
/var/log/Xorg.20.log:[130669.564] (WW) evdev: TypeMatrix.com USB Keyboard: ignoring absolute axes.
/var/log/Xorg.20.log:[146418.292] (II) evdev: TypeMatrix.com USB Keyboard: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
/var/log/Xorg.20.log:[146418.292] (II) evdev: TypeMatrix.com USB Keyboard: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
/var/log/Xorg.20.log:[146418.292] (II) evdev: 2.4G wireless USB Device 2.4G wireless USB Device: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
/var/log/Xorg.20.log:[146418.292] (II) evdev: 2.4G wireless USB Device 2.4G wireless USB Device: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
/var/log/Xorg.20.log:[146418.292] (II) evdev: HD Webcam C525: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
/var/log/Xorg.20.log:[146418.292] (II) evdev: Power Button: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
/var/log/Xorg.20.log:[146418.292] (II) evdev: Power Button: Close
/var/log/Xorg.20.log:[146418.292] (II) UnloadModule: "evdev"
```
- I use the tablet with Windows and everything works perfectly. 
- My kernel is `4.4.0-124-generic`
- The adding code seems to give the moves of the pen: for the four corners I get (approximately) 1/ `6 53093 0` 2/ `0 6 0` 3/ `95346 6 0` 4/ `95346 53639 0`

Thank you!

joseluis, 2018-05-17:
This is puzzling... So it detects the pen positioning correctly, but it doesn't seem to update the mouse cursor. And does the last number is always 0, or does it change when pressing?

I'm fearing it can be kernel related. Do you have the option of trying with a more recent kernel, like 4.13 or 4.15 ?

Unfortunately the log doesn't show anything related to the tablet... Can you please repeat the command, but this time unplugging and plugging the USB port of your tablet and waiting 10 seconds before running it?

*PS. I'm also using a typematrix 2030 keyboard, they're great!*

chupinmaxime, 2018-05-17:
I've install kernel `4.15.18-041518-generic`. This has improved the situation because now the cursor is moving... but before running your python script. 

If I execute `xinput -list`, I get the entry 
```
 ↳ NUVOTON WPM USB                         	id=13	[slave  pointer  (2)]
```
The cursor moves in my three screens. But if I run the python script, then the cursor is not moving anymore. Weirdly, the execution of the script makes the entry NUVOTON disappear if I run `xinput -list` after. 

_Yes the typematrix is a good one !_

joseluis, 2018-05-17:
What is the model of your tablet? I assumed it was GT221-PRO, but the name "NUVOTON WPM USB" gets referenced regarding the [H950P model](https://github.com/DIGImend/digimend-kernel-drivers/issues/89)...

I'm thinking that since I don't have a non-screen huion table to make tests... Maybe I should restrict the usage of these drivers to the kamvas family for now... They should work with the digimend drivers, included in the kernel.

chupinmaxime, 2018-05-18:
It is the GT221-Pro, the latest screen huion tablet.
I will install ubuntu with a virtual machine to see if it works.

chupinmaxime, 2018-05-19:
It is the GT221-Pro.
I will install ubuntu with a virtual machine to see if it works.

joseluis, 2019-03-21:
Hi, I've updated the script, adding a new debug mode to get better debugging information.

Please try the new version after changing `debug_mode = true` in the config file, and paste back some of the lines that appear when you use the pen, like the following:

```
data[12] = array('B', [8, 128, 17, 88, 33, 145, 0, 0, 0, 0, 0, 0])
{'TYPE': 128, 'X': '[8]<<16+[3]<<8+[2]=22545', 'Y': '[5]<<8+[4]=37153', 'PRESS': '[7]<<8+[6]=0'}
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22554', 'Y': '[5]<<8+[4]=37304', 'PRESS': '[7]<<8+[6]=506'}
data[12] = array('B', [8, 129, 21, 88, 54, 146, 123, 6, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22549', 'Y': '[5]<<8+[4]=37430', 'PRESS': '[7]<<8+[6]=1659'}
data[12] = array('B', [8, 129, 6, 88, 135, 146, 192, 10, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22534', 'Y': '[5]<<8+[4]=37511', 'PRESS': '[7]<<8+[6]=2752'}
```

closed 2019-04-02 by joseluis