joseluis/huion-linux-drivers #2: HUION 1060PLUS parameters

asitemade4u, 2018-05-13:
Hi @joseluis , thank you for your help on the Huion tablets.
As per your request, here are the parameters for the Huion 1060PLUS drawing tablet:

`--------------------------------------------------------------------------------
    |                Product: 1060PRO
 64 |        Params block #1: ??????
    |                          Max X: 40000
    |                          Max Y: 25000
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 79 |         Internal model: HA60-F400
 7b |         Buttons status: HK On
--------------------------------------------------------------------------------
`

joseluis, 2018-05-14:
Hi @asitemade4u thank you!

I've added the information for the tablet Huion 1060 PRO, although I've not upgrade the repository yet.

BTW: Do you have the PLUS or the PRO? how many buttons there are in your tablet? Maybe both have the same inner technical data...

asitemade4u, 2018-05-14:
Hi Jose Luis, thank you for your response.
The tablet I have is the version prior to the PRO. It is called **1060PLUS** and the reference is **X000ULLER3** (not sure this is the Serial Number or the Model Number though).
It has 6 groups of buttons on the left (each button is divided in half). I attach pictures so that you can figure it out.

asitemade4u, 2018-05-14:
BTW I get exactly the same error as described in #1 when trying to move the pen.

joseluis, 2018-05-15:
Thanks. So it's 12 buttons then.

Weirdly enough can't find the information of your tablet under a single name:

- The 1060 PLUS specs I find seems to have 5080 lines per inch, but yours seem to have 4000.
- Searching for `X000ULLER3` [google](https://www.google.com/search?q=X000ULLER3&tbm=isch) gives me mostly images associated with the 1060 Plus+ tablet. But that one apparently has 8192 pressure levels.

Please open a new issue if you wish, with the specific information of your setup, and we'll try to figure out something. Did you follow the suggestions laid out in #1 ?

I'd look into the Xorg.log for clues (`grep evdev /var/log/Xorg*.log`), as this [freedesktop bug](https://bugs.freedesktop.org/show_bug.cgi?id=104630) suggests.

Also, take a look at the issues [82](https://github.com/DIGImend/digimend-kernel-drivers/issues/82) & [77](https://github.com/DIGImend/digimend-kernel-drivers/issues/77) of the digimend-kernel-drivers repo...

asitemade4u, 2018-05-17:
I do, I am on Ubuntu 18.04 which comes with 4.15

joseluis, 2018-05-18:
I'm starting to realise that since I don't have a screenless huion tablet to make tests, I should maybe restrict these drivers to the kamvas series...

joseluis, 2019-03-21:
Hi, I've updated the script, adding a new debug mode to get better debugging information.

Please try the new version after changing `debug_mode = true` in the config file, and paste back some of the lines that appear when you use the pen, like the following:

```
data[12] = array('B', [8, 128, 17, 88, 33, 145, 0, 0, 0, 0, 0, 0])
{'TYPE': 128, 'X': '[8]<<16+[3]<<8+[2]=22545', 'Y': '[5]<<8+[4]=37153', 'PRESS': '[7]<<8+[6]=0'}
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22554', 'Y': '[5]<<8+[4]=37304', 'PRESS': '[7]<<8+[6]=506'}
data[12] = array('B', [8, 129, 21, 88, 54, 146, 123, 6, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22549', 'Y': '[5]<<8+[4]=37430', 'PRESS': '[7]<<8+[6]=1659'}
data[12] = array('B', [8, 129, 6, 88, 135, 146, 192, 10, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22534', 'Y': '[5]<<8+[4]=37511', 'PRESS': '[7]<<8+[6]=2752'}
```

joseluis, 2019-03-21:
This tablet should be [supported by the latest digimend drivers](http://digimend.github.io/tablets/Huion_New_1060_Plus/) and work out of the box with them, so I'm gonna close the issue. Feel free to reopen it if you want to keep debugging it with my driver.

closed 2019-03-21 by joseluis

acsodalatosphantom, 2019-07-17:
Can anyone pls suggest me how to set the express keys on a huion 1060 plus?