joseluis/huion-linux-drivers #5: Huion GT-185

JulianWebb, 2018-09-21:
Hardware: Inspiron 7577
Operating System: Xubuntu 18.04.1 x86_64
Console output:

```Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 64400
    |                          Max Y: 36240
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 64400
    |                          Max Y: 36240
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 c9 |      Unknown string #1: HUION_M163_161017
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name     GT-221 PRO
	Buttons               ENABLED (10)
	Scrollbar             ENABLED (1)
	Desktop notifications ENABLED
	Multi Monitor Setup   disabled

Huion Kamvas driver should now be running

% Main Menu %
button 0 = [menu_krita]
button 1 = [menu_gimp]
button 2 = [menu_darktable]
button 3 = 
button 4 = 
button 5 = 
button 6 = 
button 7 = 
button 8 = key Return
button 9 = key Escape
Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 476, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 32, in run
    main_loop()
  File "./huion-tablet-driver.py", line 268, in main_loop
    X = (data[8] << 16) + (data[3] << 8) + data[2]
IndexError: array index out of range
```
Only monitors are the laptop display and the Huion.

JulianWebb, 2018-09-22:
Some more info as I try to get it working:

Primary screen:
`eDP-1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 345mm x 194mm`

Huion screen:
`HDMI-1-4 connected 1366x768+0+0 (normal left inverted right x axis y axis) 597mm x 336mm`

The error at the bottom appears after I try to use the pen, but then on subsequent attempts to run the script it repeats the error until I modify the config, then it's back to waiting for me to try to use the pen.

As for the config, here are some changes I made:
```ini
[laptop_seperate]
# Laptop builtin and Huion tablet, separate views

# screen sizes
laptop_width = 3840
laptop_height = 2160
tablet_width = 1366
tablet_height = 768

# quick maths
total_screen_width = ${laptop_width} + ${tablet_width}
total_screen_height = ${laptop_height}
tablet_offset_x = ${laptop_width}
tablet_offset_y = 0

# output types
laptop_output = eDP-1
tablet_output = HDMI-1-4

# xrandr parameters
xrandr_laptop_output = --output ${laptop_output} --mode ${laptop_width}x${laptop_height} --pos 0x0 --rotate normal
xrandr_tablet_output = --output ${tablet_output} --mode ${tablet_width}x${tablet_height} --pos ${laptop_width}x0 --rotate normal
xrandr_args = ${xrandr_laptop_output} ${xrandr_tablet_output}
```

```ini
[tablet_gt185]
model_name    = GT-185
screen_width  = 1366
screen_height = 768
pen_max_x     = 64400
pen_max_y     = 36240
pen_max_z     = 2047
resolution    = 4000
buttons       = 8
scrollbar     = 0
screen        = true
```
I also find that the tablet is still mirroring a portion of my laptop's screen even with the script running.

JulianWebb, 2018-09-22:
I just realized that the xrandr part was not working because I forgot to enable it at the top of the config. So one problem solved.

JulianWebb, 2018-09-22:
So I've been doing a bit of digging, it seems that in the `main_loop()` you are expecting `data` to have 8 items but my tablet is only returning 7.

Here's the values I gathered from my bodge of a debug:

0. Always 7
1. Changes depending on what is happening, seems to match the other tablets.
- Default: 128
- Touch: 129
- Pen Button 1: 130
- Pen Button 2: 132
- Tablet Buttons: 224 (and sticks until touch or pen button)
2. I think this is distance the pen is from tablet but the numbers are funky, it goes past 255 and wraps as you get close
3. X value. Leftmost is 0, Rightmost is 255
4. Seems similar to #2 but I can't tell, the numbers change too fast for me
5. Y value. Topmost is 0, Bottommost is 141
6. Pressure sensitivity, 0 is min, 255 is max
7. Pressure sensitivity, 0 is min, 7 is max

I haven't tried going through the code to rematch it up yet.

JulianWebb, 2018-09-22:
I have got it working to a level I am good with and uploaded it to a new repo if you are interested in viewing:  531d55d

joseluis, 2019-03-21:
Thank you very much, this is very valuable information.

I'll have to investigate this carefully. Not having a table like this to make tests makes things harder...

JulianWebb, 2019-03-21:
I noticed in commit 1319c14, you added the tablet to the config file, however without changes to the driver script, it won't work properly.

joseluis, 2019-03-21:
Hi @JulianWebb thanks for keeping in touch.

Yes, it won't work, but I need to understand the issue much better, before commiting those kind of changes. There are more tables that don't work and I'd like to support all the cases, without causing current working ones to fail.

I've just updated the script adding a new debug mode to get better debugging information.

Could you please try the new version after changing `debug_mode = true` in the config file, and paste back some of the lines that appear when you use the pen, like the following?:

```
data[12] = array('B', [8, 128, 17, 88, 33, 145, 0, 0, 0, 0, 0, 0])
{'TYPE': 128, 'X': '[8]<<16+[3]<<8+[2]=22545', 'Y': '[5]<<8+[4]=37153', 'PRESS': '[7]<<8+[6]=0'}
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22554', 'Y': '[5]<<8+[4]=37304', 'PRESS': '[7]<<8+[6]=506'}
data[12] = array('B', [8, 129, 21, 88, 54, 146, 123, 6, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22549', 'Y': '[5]<<8+[4]=37430', 'PRESS': '[7]<<8+[6]=1659'}
data[12] = array('B', [8, 129, 6, 88, 135, 146, 192, 10, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22534', 'Y': '[5]<<8+[4]=37511', 'PRESS': '[7]<<8+[6]=2752'}
```

JulianWebb, 2019-03-21:
I will try the debug mode when I get access to my tablet later today.

joseluis, 2019-03-21:
Great. I look forward to it because I can't wrap my mind around the changes you did.

Originally the X coordinate is composed from 3 data points: `data[8]*65536 + data[3]*256 + data[2]` 
But you changed it to: `int(data[3]/255)*64400` which I don't know how it can give any meaningful coordinate representation, because if you are receiving X from data[3], by dividing between 255 and loosing the decimal part you are effectively ignoring its value completely, because the data points can't be higher than 255, at least in my tablet. So the result should be always 0.

JulianWebb, 2019-03-21:
So I got a similar error to what I originally got when I first opened this issue.
```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 64400
    |                          Max Y: 36240
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 64400
    |                          Max Y: 36240
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 c9 |       Firmware version: HUION_M163_161017
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name     GT-185
        Buttons               ENABLED (8)
        Scrollbar             disabled (0)
                 Reversed: (False)
        Desktop notifications ENABLED
        Multi Monitor Setup   disabled
        Calling xrandr        disabled

Huion Kamvas driver should now be running

% Main Menu %
button 0 = [menu_krita]
button 1 = [menu_gimp]
button 2 = [menu_darktable]
button 3 = 
button 4 = 
button 5 = [menu_vlc]
button 6 = 
button 7 = key Escape
Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 543, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 32, in run
    main_loop()
  File "./huion-tablet-driver.py", line 322, in main_loop
    X = (data[8]<<16) + (data[3]<<8) + data[2]
IndexError: array index out of range
```
I think it's probably for the same reason, the data packets coming in from my tablet are different from what it is for the others.

As for what my code is doing, I honestly had to look at again to remember.

the `int()` encapsulates the entirety of `(x_item / 255) * main.settings['pen_max_x']` so it's not losing the decimal before multiplying it, I think that was the main confusion there. I don't really know what the data packet looks like from your tablet, but I know it differs from what mine is outputting, so some tweaks had to be made.

Also I have no idea how the actual communication over USB part works, that's a mystery to me.

joseluis, 2019-03-22:
I see... Thanks. I'd love to see the received data in your case.

Could you please add the following code to your modified source, inside the main_loop() function, just before the `is_touch = data[1] == 129` line? (should be [line 236 in your fork](https://github.com/JulianWebb/huion-gt185-linux-driver/blob/master/huion-tablet-driver.py#L236) ):

```
if not (data[1] == 128):
    print("data[{}] = {}".format(len(data), data))
```

Then run it, and briefly press the four tablet corners with the pen (top left, top right, bottom left, bottom right) and paste here the output that appears when you press it?

Should give a bunch of lines similar to:
```
data[12] = array('B', [8, 128, 17, 88, 33, 145, 0, 0, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
```

joseluis, 2019-03-23:
Hey @JulianWebb I've just updated the script.

Now you can choose `[tablet_debug]` as your tablet, it should print the correct information every time you touch the screen, without giving any errors.

Let me know if you can try it and share the output

closed 2019-09-20 by joseluis