joseluis/huion-linux-drivers #6: Kamvas Pro 13

Biitblox, 2018-09-30:
Hello, I have recently got a Huion Kamvas Pro 13 and I've been looking everywhere for something that works for configuring buttons. But still there is nothing, I got the pen to work with the screen but that's about it. 

This looks really promising! Although It Did not work with the Kamvas 13. I saw that I should post this in the issues section for a module. Really looking for help, everything appreciated! ;)

(running Linux mint 19)

--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 58752
    |                          Max Y: 33048
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |      Unknown string #1: HUION_M182_180725
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------

ahloiscreamo, 2018-10-30:
Hey man,
just wondering.Are they working at all?
the pressure sensitivity / the screen? is there any wavy lines problem.?
I want to bought it, i [asked](https://twitter.com/HuionTab/status/1055628416706203648) Huion on Twitter and they replied Huion Kamvas Pro 13 working on linux.
If they work minus the button it would be great news for me because i don't use button but i prefer keyboard.
Thanks mate

Biitblox, 2018-11-01:
Yeah the screen works great! I have not experienced any wavy lines what so ever using krita.
The only down side about the tablet is size but that's not so bad. I use this tablet mostly for sculpting and at first extremely wanted the button functionality but now I found a workflow that is fine with the keyboard.

(Not that having buttons would be great though.)

wwklnd, 2019-01-02:
I have a Kamvas Pro 13 as well that I'm trying to get working, and currently only the screen seems to function (in Manjaro). When I try to run the driver file, I get this output:
`$ sudo ./huion-tablet-driver.py
[sudo] password for william: 
Finding USB device. . . Done!
grabbed interface %d 0
grabbed interface %d 1
Reading configuration. . . Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 476, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 27, in run
    read_config()
  File "./huion-tablet-driver.py", line 410, in read_config
    'total_screen_width').split("#",1)[0].strip())
  File "/usr/lib/python3.7/ast.py", line 91, in literal_eval
    return _convert(node_or_string)
  File "/usr/lib/python3.7/ast.py", line 90, in _convert
    return _convert_signed_num(node)
  File "/usr/lib/python3.7/ast.py", line 63, in _convert_signed_num
    return _convert_num(node)
  File "/usr/lib/python3.7/ast.py", line 55, in _convert_num
    raise ValueError('malformed node or string: ' + repr(node))
ValueError: malformed node or string: <_ast.BinOp object at 0x7fb86db10320>
`

I feel way out of my depth here, anyone got ideas?

tylerarizzo, 2019-01-21:
I also have a Kamvas Pro 13 (GT-133) and I cannot get any aspect of it working.  I can see the data of the pen moving over the screen using this https://github.com/benthor/HuionKamvasGT191LinuxDriver, but the cursor does not move.  Here is my output running this driver:

>Finding USB device. . . Done!
Reading configuration. . . Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 476, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 27, in run
    read_config()
  File "./huion-tablet-driver.py", line 419, in read_config
    'xrandr_args').split("#",1)[0].strip()
  File "/usr/lib/python3.6/configparser.py", line 800, in get
    d)
  File "/usr/lib/python3.6/configparser.py", line 455, in before_get
    self._interpolate_some(parser, option, L, value, section, defaults, 1)
  File "/usr/lib/python3.6/configparser.py", line 507, in _interpolate_some
    option, section, rawval, ":".join(path)) from None
configparser.InterpolationMissingOptionError: Bad value substitution: option 'xrandr_args' in section 'monitor_2' contains an interpolation key 'xrandr_output2' which is not a valid option name. Raw value: '${xrandr_output1} ${xrandr_output2}'

asdzocker, 2019-02-01:
Hi there, also trying to use a GT-133 on Ubuntu 18.04, but I can't get the pen working. Driver output states this:
`Finding USB device. . . Done!
Reading configuration. . . Traceback (most recent call last):
  File "/usr/lib/python3.6/configparser.py", line 1138, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'tablet_gt133'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 476, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 27, in run
    read_config()
  File "./huion-tablet-driver.py", line 377, in read_config
    main.settings['model_name'] = config.get(current_tablet, 'model_name')
  File "/usr/lib/python3.6/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.6/configparser.py", line 1141, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'tablet_gt133'
`
Any ideas what to do?

Gijsjan, 2019-02-22:
I am not a Python dev, but all errors are 'simple' bugs. If you edit the Python-script you can resolve them.
@wwklnd ast.literal_eval will not parse an operator. I replace it with `eval` and that works
@tylerarizzo the monitor_2 setup misses a var: `xrandr_output3` that should be `xrandr_output2`
@asdzocker the config.ini does not specify a config for `tablet_g133`, so it can not find that var

asdzocker, 2019-02-22:
Yeah, I already figured that far. I don't know why, but for some obscure reason, it seems to work right now. I bought an USB-Docking station and plugged it into that, and now it just works. But now I encountered another problem, not as important but nevertheless most annoying. I have a set up with three displays, including the tablet itself. The input is mapped to all screens, which makes it hard to draw straight lines, since the scaling is all messed up (which was to be expected, I've configured two screens side by side and the tablet in lower right corner). Is there a way to map the input to one screen, preferably the tablet?

asdzocker, 2019-03-17:
alright, resolved the problems regarding dependencies, discovering that uclogic-tools was located at / rather than in usr/local/bin as specified in the config.ini. Solved that by moving the directory and correcting the path. Also, I did include the GT-133 and its corresponding values in the config.ini, as specified by uclogic. Having that fixed, I encountered a new problem, I'm unable to solve: while trying to run the driver.py I get a KeyError: 'config' referring to the configparser.py at line 1138. complete output can be found below.
`sudo ./Fremdpakete/Huion-bearb/huion-tablet-driver.py 
Finding USB device. . . Done!
Reading configuration. . . ConfigParser initalisiert... Reading config.ini... Traceback (most recent call last):
  File "/usr/lib/python3.6/configparser.py", line 1138, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'config'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./Fremdpakete/Huion-bearb/huion-tablet-driver.py", line 479, in <module>
    main.run()
  File "./Fremdpakete/Huion-bearb/huion-tablet-driver.py", line 27, in run
    read_config()
  File "./Fremdpakete/Huion-bearb/huion-tablet-driver.py", line 378, in read_config
    'current_tablet').split("#",1)[0].strip('[]').strip()
  File "/usr/lib/python3.6/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.6/configparser.py", line 1141, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'config'
`
Changes I made to the config.ini
`[tablet_gt133]
model_name    = GT-133
screen_width  = 1920
screen_height = 1080
pen_max_x     = 58752
pen_max_y     = 33048
pen_max_z     = 8191
resolution    = 5080
buttons       = 5
scrollbar     = 1
screen        = true`

Output by uclogic-probe | uclogic decode:
`   |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 58752
    |                          Max Y: 33048
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M182_181019
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
`
Hoping that anyone can help me with that, I'm done with my (not so extensive) python knowledge.

joseluis, 2019-03-21:
Hi, I've updated the script, adding a new debug mode to get better debugging information.

Please try the new version after changing `debug_mode = true` in the config file, and paste back some of the lines that appear when you use the pen, like the following:

```
{'TYPE': 128, 'X': '[8]<<16+[3]<<8+[2]=22545', 'Y': '[5]<<8+[4]=37153', 'PRESS': '[7]<<8+[6]=0'}
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22554', 'Y': '[5]<<8+[4]=37304', 'PRESS': '[7]<<8+[6]=506'}
data[12] = array('B', [8, 129, 21, 88, 54, 146, 123, 6, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22549', 'Y': '[5]<<8+[4]=37430', 'PRESS': '[7]<<8+[6]=1659'}
data[12] = array('B', [8, 129, 6, 88, 135, 146, 192, 10, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22534', 'Y': '[5]<<8+[4]=37511', 'PRESS': '[7]<<8+[6]=2752'}
data[12] = array('B', [8, 129, 245, 87, 183, 146, 234, 13, 0, 0, 0, 0])
```

asdzocker, 2019-03-24:
While trying to install the driver, I encountered another problem, which I neither know origin nor fix for, due to my unability to find the returned error code in the texts. After completing reading the config, this appears:

> /bin/sh: 1: /usr/local/bin/uclogic-probe: not found
> /bin/sh: 1: /usr/local/bin/uclogic-decode: not found
> Preparing driver. . . ERROR running the following comand:
>     "/usr/local/bin//uclogic-probe" "3" "9" | "/usr/local/bin//uclogic-decode"
> RETURN CODE: 127 

I do know for sure, that both uclogic-probe and uclogic-decode are located at this directory, I moved them there, next to everything else contained in it's folder. As said above, I've tried to find out, what causes something returning 127 or what this even means. uclogic itself works, I've also tested that with the given (and correct) parameters.

joseluis, 2019-03-24:
Error 127 is just the shell error for "command not found".

If you had really copied the files there manually, but lacking the right permissions, the error code could be 126. But since it's 127, that means the files are not really where you think they are.

If you run the command:

```
ls -l /usr/local/bin/uclogic-*
```

It should show you the files like this (if not, that means they're really not there, and maybe you moved the container folder instead of those files directly, or something like that):
```
-rwxr-xr-x 1 root root 36856 mar 21 15:15 /usr/local/bin/uclogic-decode
-rwxr-xr-x 1 root root 26744 mar 21 15:15 /usr/local/bin/uclogic-probe
```

To avoid errors, you should [compile and install the uclogic-tools the proper way](https://github.com/joseluis/huion-linux-drivers/#dependencies):

```
$ sudo apt install make automake gcc pkg-config libusb-1.0-0-dev

$ git clone https://github.com/DIGImend/uclogic-tools
$ cd uclogic-tools
$ autoreconf -i -f && ./configure --prefix=/usr/local/ && make
$ sudo make install
```

asdzocker, 2019-03-24:
Alright, after removing and reinstalling uclogic in the correct way, it does seem to work, up to the point, where the pen comes into play:

> Setting up driver. . . Done!
> 	Tablet model name     GT-133
> 	Buttons               ENABLED (5)
> 	Scrollbar             ENABLED (1)
> 	Desktop notifications ENABLED
> 	Multi Monitor Setup   ENABLED
> unable to find device Tablet Monitor Pen 164942
> Setting up multiple monitors. . . ERROR running the following comand:
> 	xinput set-prop "Tablet Monitor Pen 164942" --type=float "Coordinate Transformation Matrix" 0.3 0 0.7 0 0.75 0.0 0 0 1
> RETURN CODE: 1

Any idea how to fix this? Is there anything I should have done before or something?

joseluis, 2019-03-24:
Hmmm... My guess is that you forgot to do the [Xorg extra code](https://github.com/joseluis/huion-linux-drivers/#xorg-extra-code) part.

Apart from that, I updated the script a lot the last couple of days. so better use the last version available of both script and config.

Then in the config.ini file set either `debug_mode = true`, or `current_tablet = [tablet_debug]`. Run the driver, touch the screen in corners as indicated and paste here the output, so I can have more complete info for your tablet to try and update the script to support it better.

Apart from that, if it keeps failing in the "Setting up multiple monitors" you could set `enable_multi_monitor = false` for now, and see if it runs enough to print the debug info.

asdzocker, 2019-03-27:
I tried to do as specified in the Xorg guide. but unfortunately, I don't have the directory ` /etc/X11/xorg.conf.d/` instead, I've got a directory with that name under` usr/share/X11` what am I supposed to do now? Since I learned earlier, that simply moving or copying the directory probably won't work.
I'm using the latest version of your script, as recommended.

asdzocker, 2019-03-27:
So, I learned,that a part of the problem was my i-tec USB 3.0 Docking Station. Since I put the USB-plug direct to my laptop the pen is there and works. Now I did the calibration-mod and this was the output
`Huion Kamvas driver should now be running

Please slowly and briefly touch the LEFT UP corner of your tablet:
...
data[12] = array('B', [8, 129, 203, 2, 37, 4, 13, 0, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 201, 2, 42, 4, 186, 0, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 199, 2, 47, 4, 121, 1, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 197, 2, 49, 4, 52, 2, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 199, 2, 49, 4, 214, 2, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 202, 2, 49, 4, 88, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 206, 2, 48, 4, 185, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 208, 2, 46, 4, 253, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 209, 2, 42, 4, 56, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 211, 2, 41, 4, 108, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 212, 2, 41, 4, 162, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 211, 2, 43, 4, 214, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 209, 2, 46, 4, 6, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 209, 2, 47, 4, 55, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 212, 2, 47, 4, 98, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 44, 4, 142, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 44, 4, 177, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 42, 4, 208, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 212, 2, 40, 4, 237, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 212, 2, 35, 4, 21, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 31, 4, 62, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 215, 2, 29, 4, 98, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 28, 4, 124, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 28, 4, 146, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 212, 2, 29, 4, 159, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 211, 2, 32, 4, 171, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 210, 2, 34, 4, 181, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 210, 2, 35, 4, 198, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 211, 2, 36, 4, 233, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 210, 2, 39, 4, 33, 7, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 207, 2, 41, 4, 105, 7, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 205, 2, 43, 4, 169, 7, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 206, 2, 44, 4, 244, 7, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 208, 2, 47, 4, 56, 8, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 211, 2, 49, 4, 114, 8, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 50, 4, 154, 8, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 50, 4, 188, 8, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 50, 4, 226, 8, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 49, 4, 7, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 49, 4, 43, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 215, 2, 48, 4, 93, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 217, 2, 48, 4, 149, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 217, 2, 48, 4, 197, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 215, 2, 48, 4, 246, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 213, 2, 49, 4, 26, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 47, 4, 58, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 217, 2, 46, 4, 81, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 219, 2, 43, 4, 104, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 221, 2, 42, 4, 124, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 221, 2, 40, 4, 134, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 222, 2, 40, 4, 140, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 224, 2, 41, 4, 150, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 226, 2, 40, 4, 155, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 226, 2, 39, 4, 155, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 224, 2, 38, 4, 148, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 219, 2, 38, 4, 145, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 2, 39, 4, 151, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 209, 2, 41, 4, 155, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 207, 2, 45, 4, 150, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 207, 2, 48, 4, 144, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 205, 2, 53, 4, 138, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 202, 2, 58, 4, 120, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 199, 2, 62, 4, 97, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 199, 2, 68, 4, 56, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 197, 2, 72, 4, 9, 10, 0, 0, 0, 0])
[... 145 lines cut ...]
data[12] = array('B', [8, 129, 236, 226, 218, 126, 136, 5, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 239, 226, 218, 126, 181, 5, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 241, 226, 218, 126, 222, 5, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 243, 226, 218, 126, 2, 6, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 244, 226, 219, 126, 29, 6, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 245, 226, 219, 126, 50, 6, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 246, 226, 219, 126, 64, 6, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 246, 226, 219, 126, 72, 6, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 246, 226, 219, 126, 74, 6, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 245, 226, 219, 126, 67, 6, 0, 0, 235, 250])
[... cut ...]
data[12] = array('B', [8, 129, 216, 226, 219, 126, 71, 4, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 216, 226, 219, 126, 97, 3, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 216, 226, 219, 126, 104, 2, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 216, 226, 219, 126, 114, 1, 0, 0, 235, 250])
data[12] = array('B', [8, 129, 216, 226, 219, 126, 141, 0, 0, 0, 235, 250])
...
data[12] = array('B', [8, 0, 140, 227, 250, 118, 0, 0, 0, 0, 235, 250])`
There were no admission for the left down corner.
I'm with the pen nearly 15 cm to high, I guess.

joseluis, 2019-03-27:
> There were no admission for the left down corner.

Thanks. Fixed

Your values seem OK. These are the values of corners LEFT_UP and RIGHT_DOWN translated to coordinates. It is consistent with the `pen_max_x` and `pen_max_y` parameters. I guess your tablet is working well, right?

```
Please slowly and briefly touch the LEFT UP corner of your tablet:
...
{'X': '715', 'Y': '1061', 'PRESS': '13'}
{'X': '713', 'Y': '1066', 'PRESS': '186'}
{'X': '711', 'Y': '1071', 'PRESS': '377'}
{'X': '709', 'Y': '1073', 'PRESS': '564'}
{'X': '711', 'Y': '1073', 'PRESS': '726'}
{'X': '714', 'Y': '1073', 'PRESS': '856'}
{'X': '718', 'Y': '1072', 'PRESS': '953'}
{'X': '720', 'Y': '1070', 'PRESS': '1021'}
{'X': '721', 'Y': '1066', 'PRESS': '1080'}
{'X': '723', 'Y': '1065', 'PRESS': '1132'}
{'X': '724', 'Y': '1065', 'PRESS': '1186'}
{'X': '723', 'Y': '1067', 'PRESS': '1238'}
{'X': '721', 'Y': '1070', 'PRESS': '1286'}
{'X': '721', 'Y': '1071', 'PRESS': '1335'}
{'X': '724', 'Y': '1071', 'PRESS': '1378'}
{'X': '726', 'Y': '1068', 'PRESS': '1422'}
{'X': '726', 'Y': '1068', 'PRESS': '1457'}
{'X': '725', 'Y': '1066', 'PRESS': '1488'}
{'X': '724', 'Y': '1064', 'PRESS': '1517'}
{'X': '724', 'Y': '1059', 'PRESS': '1557'}
{'X': '725', 'Y': '1055', 'PRESS': '1598'}
{'X': '727', 'Y': '1053', 'PRESS': '1634'}
{'X': '726', 'Y': '1052', 'PRESS': '1660'}
{'X': '725', 'Y': '1052', 'PRESS': '1682'}
{'X': '724', 'Y': '1053', 'PRESS': '1695'}
{'X': '723', 'Y': '1056', 'PRESS': '1707'}
{'X': '722', 'Y': '1058', 'PRESS': '1717'}
{'X': '722', 'Y': '1059', 'PRESS': '1734'}
{'X': '723', 'Y': '1060', 'PRESS': '1769'}
{'X': '722', 'Y': '1063', 'PRESS': '1825'}
{'X': '719', 'Y': '1065', 'PRESS': '1897'}
{'X': '717', 'Y': '1067', 'PRESS': '1961'}
{'X': '718', 'Y': '1068', 'PRESS': '2036'}
{'X': '720', 'Y': '1071', 'PRESS': '2104'}
{'X': '723', 'Y': '1073', 'PRESS': '2162'}
{'X': '725', 'Y': '1074', 'PRESS': '2202'}
{'X': '726', 'Y': '1074', 'PRESS': '2236'}
{'X': '726', 'Y': '1074', 'PRESS': '2274'}
{'X': '725', 'Y': '1073', 'PRESS': '2311'}
{'X': '725', 'Y': '1073', 'PRESS': '2347'}
{'X': '727', 'Y': '1072', 'PRESS': '2397'}
{'X': '729', 'Y': '1072', 'PRESS': '2453'}
{'X': '729', 'Y': '1072', 'PRESS': '2501'}
{'X': '727', 'Y': '1072', 'PRESS': '2550'}
{'X': '725', 'Y': '1073', 'PRESS': '2586'}
{'X': '726', 'Y': '1071', 'PRESS': '2618'}
{'X': '729', 'Y': '1070', 'PRESS': '2641'}
{'X': '731', 'Y': '1067', 'PRESS': '2664'}
{'X': '733', 'Y': '1066', 'PRESS': '2684'}
{'X': '733', 'Y': '1064', 'PRESS': '2694'}
{'X': '734', 'Y': '1064', 'PRESS': '2700'}
{'X': '736', 'Y': '1065', 'PRESS': '2710'}
{'X': '738', 'Y': '1064', 'PRESS': '2715'}
{'X': '738', 'Y': '1063', 'PRESS': '2715'}
{'X': '736', 'Y': '1062', 'PRESS': '2708'}
{'X': '731', 'Y': '1062', 'PRESS': '2705'}
{'X': '726', 'Y': '1063', 'PRESS': '2711'}
{'X': '721', 'Y': '1065', 'PRESS': '2715'}
{'X': '719', 'Y': '1069', 'PRESS': '2710'}
{'X': '719', 'Y': '1072', 'PRESS': '2704'}
{'X': '717', 'Y': '1077', 'PRESS': '2698'}
[... 64 lines cut ...]
{'X': '58101', 'Y': '32475', 'PRESS': '1586'}
{'X': '58102', 'Y': '32475', 'PRESS': '1600'}
{'X': '58102', 'Y': '32475', 'PRESS': '1608'}
{'X': '58102', 'Y': '32475', 'PRESS': '1610'}
{'X': '58101', 'Y': '32475', 'PRESS': '1603'}
{'X': '58099', 'Y': '32475', 'PRESS': '1589'}
{'X': '58097', 'Y': '32475', 'PRESS': '1565'}
{'X': '58093', 'Y': '32475', 'PRESS': '1528'}
{'X': '58089', 'Y': '32475', 'PRESS': '1474'}
{'X': '58084', 'Y': '32475', 'PRESS': '1396'}
{'X': '58078', 'Y': '32475', 'PRESS': '1278'}
{'X': '58072', 'Y': '32475', 'PRESS': '1095'}
{'X': '58072', 'Y': '32475', 'PRESS': '865'}
{'X': '58072', 'Y': '32475', 'PRESS': '616'}
{'X': '58072', 'Y': '32475', 'PRESS': '370'}
{'X': '58072', 'Y': '32475', 'PRESS': '141'}
```

> I'm with the pen nearly 15 cm to high, I guess.

Do you mean that it is not calibrated?

You could configure a recalibration with `enable_calibration = true` and playing with the calibrate_min_ and calibrate_max_` parameters. Adding to the `min`, from 0,  and substracting from the `max`, from your tablet's `pen_max_` in order to compensate wrong consistent displacements.

Please let me know if everything is OK.

asdzocker, 2019-04-05:
Alright, did as you recommended and played with the calibration data. But appearently I can't figure the right ones out. For my understanding: x-axis is Left to right and y-axis is up-down, right? And the calibration values alter the area, in which the pen is registered? Or do they change the area, where the mouse is to be placed, when getting interaction with the pen? 

What I want the pen to do is staying on the screen of the tablet, when I'm using it and not messing around on one of the other two screens above. Is Playing with calibration values the right way to do so, or should I enable Multi-monitor in the config.ini and change something there?

joseluis, 2019-04-05:
Oh then yes, in that case absolutely, you need to set `enable_multi_monitor = true` and don'tmmess with calibration for now.

If you have 2 screens then set `current_monitor_setup = [monitor_2]` and set the appropriate screens widths and heights for both of them in that section (second screen would be the tablet, arranged left to right)

joseluis, 2019-04-28:
I'm closing this. But please feel free to reopen if you need it.

closed 2019-04-28 by joseluis

choller, 2019-08-28:
> So, I learned,that a part of the problem was my i-tec USB 3.0 Docking Station. 

Fwiw, I had the exact same problem with a USB-C docking station. The tablet just wouldn't provide any inputs, despite all my debugging efforts. Plugging the USB cable directly into the laptop solved all problems. Noting this in FAQ/README could save some people a lot of time. Thanks for working on this!