joseluis/huion-linux-drivers #8: GT-221 Pro - Tablet does not respond to input.

bananapotsticker, 2018-11-26:
Hello!

Just for quick reference: I'm running KDE neon 5.14 with the x86_64 Linux 4.15.0-39-generic kernel and am attempting to get my GT-221 Pro working under it.

I've been struggling to get my GT-221 Pro working under linux and I've exhausted just about everything I could try. Basically no matter how I set up my config.ini it either spits out an error (because I think I'm setting the monitor setup wrong for multi-monitor) or it runs, but I can't get the tablet to respond to either pen input or by pressing any buttons. Also, yes, I have all my dependencies installed and yes, I added the /etc/X11/xorg.conf.d/evdev-tablet.conf

**FYI: When I try to use one of the pre-made monitor configurations (or even not bother with multi-monitor at all and only have the tablet plugged in,  the driver 'works' but the tablet still refuses to notice the pen or when I press any buttons.**

Anyway. I'll attach the full config.ini but here's the part I wrote myself.

```

[monitor_4]
# 2 monitors arranged vertically with the tablet being the lower one.

screen_1W = 1920
screen_1H = 1080
screen_2W = 1920
screen_2H = 1080

total_screen_width   = ${screen_1W}
total_screen_height  = ${screen_1H} + ${screen_2H}
tablet_offset_x      = 0
tablet_offset_y      = {screen_1H}

xrandr_output1 = --output HDMI-A-0 --primary --mode 1920x1080 --pos 0x1080 --rotate normal
xrandr_output22 = --output DVI-D-0 --mode 1920x1080 --pos 0x0 --rotate normal
xrandr_args = ${xrandr_output1} ${xrandr_output2}

```

Here's the information for my monitors that I got from arandr:

```
#!/bin/sh
xrandr --output DisplayPort-0 --off --output DVI-D-0 --mode 1920x1080 --pos 0x0 --rotate normal --output DVI-D-1 --off --output HDMI-A-0 --primary --mode 1920x1080 --pos 0x1080 --rotate normal

```

Here's the output I get when I try to use [monitor_4]:

```
den@Ignition2:~/huion-linux-drivers$ sudo ./huion-tablet-driver.py
Finding USB device. . . Done!
Reading configuration. . . Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 476, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 27, in run
    read_config()
  File "./huion-tablet-driver.py", line 419, in read_config
    'xrandr_args').split("#",1)[0].strip()
  File "/usr/lib/python3.6/configparser.py", line 800, in get
    d)
  File "/usr/lib/python3.6/configparser.py", line 455, in before_get
    self._interpolate_some(parser, option, L, value, section, defaults, 1)
  File "/usr/lib/python3.6/configparser.py", line 507, in _interpolate_some
    option, section, rawval, ":".join(path)) from None
configparser.InterpolationMissingOptionError: Bad value substitution: option 'xrandr_args' in section 'monitor_2' contains an interpolation key 'xrandr_output2' which is not a valid option name. Raw value: '${xrandr_output1} ${xrandr_output2}'

```

And here's the output when the driver successfully runs (either from not using multimonitor or by using one of the pre-made configurations.)
```

den@Ignition2:~/huion-linux-drivers$ sudo ./huion-tablet-driver.py 
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 37540
    |                          Max Y: 21120
    |                   Max pressure: 8191
    |                     Resolution: 2000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 95352
    |                          Max Y: 53645
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M167_170623
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name     GT-221 PRO
        Buttons               ENABLED (10)
        Scrollbar             ENABLED (1)
        Desktop notifications ENABLED
        Multi Monitor Setup   disabled

Huion Kamvas driver should now be running

% Main Menu %
button 0 = [menu_krita]
button 1 = [menu_gimp]
button 2 = [menu_darktable]
button 3 = 
button 4 = 
button 5 = 
button 6 = 
button 7 = 
button 8 = key Return
button 9 = key Escape

```

Thanks for any help you can provide.

I'd like to note that using https://github.com/joseluis/HuionKamvasGT221PROLinuxDriver **works**, but I get **no pen pressure**.

[arandr_output.txt](https://github.com/joseluis/huion-linux-drivers/files/2616768/arandr_output.txt)
[config.txt](https://github.com/joseluis/huion-linux-drivers/files/2616770/config.txt)

Gijsjan, 2019-02-22:
A variable is missing: see this pull request: https://github.com/joseluis/huion-linux-drivers/pull/12

joseluis, 2019-03-21:
Hi, I've updated the script, adding a new debug mode to get better debugging information.

Please try the new version after changing `debug_mode = true` in the config file, and paste back some of the lines that appear when you use the pen, like the following:

```
data[12] = array('B', [8, 128, 17, 88, 33, 145, 0, 0, 0, 0, 0, 0])
{'TYPE': 128, 'X': '[8]<<16+[3]<<8+[2]=22545', 'Y': '[5]<<8+[4]=37153', 'PRESS': '[7]<<8+[6]=0'}
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22554', 'Y': '[5]<<8+[4]=37304', 'PRESS': '[7]<<8+[6]=506'}
data[12] = array('B', [8, 129, 21, 88, 54, 146, 123, 6, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22549', 'Y': '[5]<<8+[4]=37430', 'PRESS': '[7]<<8+[6]=1659'}
data[12] = array('B', [8, 129, 6, 88, 135, 146, 192, 10, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22534', 'Y': '[5]<<8+[4]=37511', 'PRESS': '[7]<<8+[6]=2752'}
```

closed 2019-04-02 by joseluis