joseluis/huion-linux-drivers #9: Support for GT-156HD V2

cardboardcode, 2018-12-03:
Hi @joseluis, thank you for the hard work you have put into this repository. 

The tablet model I am attempting to run is **GT-156HD V2** and attached below is the console output that appears just after executing the driver.

```
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 54200
    |                          Max Y: 30480
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 68834
    |                          Max Y: 38710
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M174_170925
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
``` 
After installing the dependencies and running **sudo ./huion-tablet-driver.py**, the tablet model is detected to be **GT-221 Pro**. The tablet pen is unable to move the cursor on the tablet monitor but the pressure input seems to be working. **In other words, I can select with the tablet but I cannot move the cursor at all.**

Please help.

Feel free to ask if you require more information. Thank you.

joseluis, 2019-03-21:
Hi, I've updated the script, adding a new debug mode to get better debugging information.

Please try the new version after changing `debug_mode = true` in the config file, and paste back some of the lines that appear when you use the pen, like the following:

```
data[12] = array('B', [8, 128, 17, 88, 33, 145, 0, 0, 0, 0, 0, 0])
{'TYPE': 128, 'X': '[8]<<16+[3]<<8+[2]=22545', 'Y': '[5]<<8+[4]=37153', 'PRESS': '[7]<<8+[6]=0'}
data[12] = array('B', [8, 129, 26, 88, 184, 145, 250, 1, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22554', 'Y': '[5]<<8+[4]=37304', 'PRESS': '[7]<<8+[6]=506'}
data[12] = array('B', [8, 129, 21, 88, 54, 146, 123, 6, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22549', 'Y': '[5]<<8+[4]=37430', 'PRESS': '[7]<<8+[6]=1659'}
data[12] = array('B', [8, 129, 6, 88, 135, 146, 192, 10, 0, 0, 0, 0])
{'TYPE': 129, 'X': '[8]<<16+[3]<<8+[2]=22534', 'Y': '[5]<<8+[4]=37511', 'PRESS': '[7]<<8+[6]=2752'}
```

the-real-a99d, 2019-08-08:
Hi @joseluis, I've tried your driver, the buttons and the scrollbar seems to work but the pen doesn't output anything. (mint-user)

there is my output when I start the script:

Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 54200
    |                          Max Y: 30480
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 68834
    |                          Max Y: 38710
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M174_170925
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name         GT-156 v2
	Buttons                   ENABLED (14)
	Scrollbar                 ENABLED (1)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    Enabled (1920x1080)
	Current Monitor Setup     [monitor_3]
	Multi Monitor Setup       ENABLED
	Calling xrandr            disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.15.0-55-generic (x86_64 x86_64)
#60-Ubuntu SMP Tue Jul 2 18:22:20 UTC 2019

DEVICE: /dev/input/event4 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=68834, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=38710, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]
VPEN:
name "Tablet Monitor Pen 133801", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

Setting up multiple monitors. . . 
Running xinput. . . 
» xinput set-prop "Tablet Monitor Pen 133801" --type=float "Coordinate Transformation Matrix" 0.3333333333333333 0 0.6666666666666666 0 1.0 0.0 0 0 1
Mapped tablet area to "1920x1080 + 3840x0"

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

[menu_simple_14b]
button 0 = key Tab           # hide interface
button 1 = key r             # rect select (gimp) & pick layer (krita)
button 2 = key ctrl+x        # cut
button 3 = key ctrl+c        # copy
button 4 = key ctrl+v        # paste
button 5 = key ctrl+z        # undo
button 6 = key ctrl+y        # redo (gimp)
button 7 = key ctrl+shift+z  # redo (krita)
button 8 = key 4            # turn left (krita)
button 9 = key 6            # turn right (krita)
button 10 = 
button 11 = 
button 12 = 
button 13 =

closed 2019-08-10 by joseluis