joseluis/huion-linux-drivers #11: Huion Kamvas GT 221 Pro on Linux Mint 19.1 not running

DrHoofman, 2019-02-11:
Hi @joseluis,
I try to setup my recently bought Huion Kamvas GT 221 Pro on Linux Mint 19.1 but I can't get it running properly.
Sorry, I'm far away of being a programmer  
 maybe you can help me..
I have a 3 Monitor Setup

<stefan@stefan-Precision-M6700:~$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Tablet Monitor Pen                      	id=11	[slave  pointer  (2)]
⎜   ↳ Logitech M510                           	id=13	[slave  pointer  (2)]
⎜   ↳ Logitech K540/K545                      	id=14	[slave  pointer  (2)]
⎜   ↳ AlpsPS/2 ALPS DualPoint Stick           	id=18	[slave  pointer  (2)]
⎜   ↳ AlpsPS/2 ALPS DualPoint TouchPad        	id=19	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Tablet Monitor Pad                      	id=12	[slave  keyboard (3)]
    ↳ Laptop_Integrated_Webcam_E4HD:          	id=15	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=16	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=17	[slave  keyboard (3)]
    ↳ Logitech K540/K545                      	id=20	[slave  keyboard (3)]
stefan@stefan-Precision-M6700:~$ grep evdev /var/log/Xorg*.log | tail -20
[  3060.351] (II) Using input driver 'evdev' for 'Tablet Monitor Pad'
[  3060.351] (**) evdev: Tablet Monitor Pad: Device: "/dev/input/event13"
[  3060.408] (--) evdev: Tablet Monitor Pad: Vendor 0x256c Product 0x6e
[  3060.408] (--) evdev: Tablet Monitor Pad: Found 12 mouse buttons
[  3060.408] (II) evdev: Tablet Monitor Pad: Forcing relative x/y axes to exist.
[  3060.408] (II) evdev: Tablet Monitor Pad: Configuring as mouse
[  3060.408] (**) evdev: Tablet Monitor Pad: YAxisMapping: buttons 4 and 5
[  3060.408] (**) evdev: Tablet Monitor Pad: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[  3060.412] (**) Tablet Monitor Pen: Applying InputClass "evdev tablet catchall"
[  3060.412] (**) Tablet Monitor Pen: Applying InputClass "evdev tablet catchall"
[  3060.412] (II) Using input driver 'evdev' for 'Tablet Monitor Pen'
[  3060.412] (**) evdev: Tablet Monitor Pen: Device: "/dev/input/event12"
[  3060.412] (--) evdev: Tablet Monitor Pen: Vendor 0x256c Product 0x6e
[  3060.412] (--) evdev: Tablet Monitor Pen: Found absolute axes
[  3060.412] (--) evdev: Tablet Monitor Pen: Found x and y absolute axes
[  3060.412] (--) evdev: Tablet Monitor Pen: Found absolute tablet.
[  3060.412] (II) evdev: Tablet Monitor Pen: Configuring as tablet
[  3060.412] (**) evdev: Tablet Monitor Pen: YAxisMapping: buttons 4 and 5
[  3060.412] (**) evdev: Tablet Monitor Pen: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[  3060.412] (II) evdev: Tablet Monitor Pen: initialized for absolute axes.>

I downloaded your code, installed the dependecies and the extra code but get the following:

<stefan@stefan-Precision-M6700:~$ sudo ./huion-tablet-driver.py
[sudo] Passwort für stefan:          
sudo: ./huion-tablet-driver.py: Befehl nicht gefunden>

I have no clue whats going on..

Please tell me what information you need and how to get them..

Thank you in advance
Stefan

DrHoofman, 2019-02-11:
oops made something wrong, I didn't want to cancel these quotes...

joseluis, 2019-03-21:
Hi @DrHoofman 

That error seems to mean "command not found" in german.

I see you are trying to execute the command from your home folder. But if the script is not there (and it shouldn't be there) it wont work.

You must first enter the directory where the script is, and then the command will work. I don't know where you downloaded the script. But this is a somewhat complete example that should work:

```
sudo apt install git
git clone https://github.com/joseluis/huion-linux-drivers.git
cd huion-linux-driver
sudo ./huion-tablet-driver.py
```

P.S. I surrounded your terminal output with code quotes ` ``` ``` ` to avoid the formatting issues

closed 2019-04-02 by joseluis