joseluis/huion-linux-drivers #15: Huion gt191 missing tablet capabilities: resolution.

kalle-ankan, 2019-03-24:
Hello

We have an HUION GT-191 that is not working.

We try to use it on Debian buster, below have I listed some output.
It seems that  the input driver is missing tablet capabilities: resolution.
Any idea how debug it or make it working?

lsub:
lsusb -v -d  256c:006e

Bus 002 Device 009: ID 256c:006e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x256c 
  idProduct          0x006e 
  bcdDevice            0.00
  iManufacturer           5 (error)
  iProduct                6 (error)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              480mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      83
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2

xinput -list

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SONiX USB Device Mouse                  	id=10	[slave  pointer  (2)]
⎜   ↳ Telink Wireless Receiver Mouse          	id=13	[slave  pointer  (2)]
⎜   ↳ Telink Wireless Receiver Consumer Control	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0808)                  	id=8	[slave  keyboard (3)]
    ↳ SONiX USB Device                        	id=9	[slave  keyboard (3)]
    ↳ SONiX USB Device System Control         	id=11	[slave  keyboard (3)]
    ↳ Telink Wireless Receiver                	id=12	[slave  keyboard (3)]
    ↳ Telink Wireless Receiver System Control 	id=15	[slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                      	id=16	[slave  keyboard (3)]
    ↳ Telink Wireless Receiver Consumer Control	id=17	[slave  keyboard (3)]

And then xorg log file:
[    50.704] (II) config/udev: Adding input device Tablet Monitor Pen 172738 (/dev/input/event23)
[    50.704] (**) Tablet Monitor Pen 172738: Applying InputClass "evdev tablet catchall"
[    50.704] (**) Tablet Monitor Pen 172738: Applying InputClass "libinput tablet catchall"
[    50.704] (II) Using input driver 'libinput' for 'Tablet Monitor Pen 172738'
[    50.704] (**) Tablet Monitor Pen 172738: always reports core events
[    50.705] (**) Option "Device" "/dev/input/event23"
[    50.705] (**) Option "_source" "server/udev"
[    50.706] (II) event23 - Tablet Monitor Pen 172738: is tagged by udev as: Tablet
[    50.706] (EE) event23 - Tablet Monitor Pen 172738: libinput bug: missing tablet capabilities: resolution. Ignoring this device.
[    50.706] (II) event23 - Tablet Monitor Pen 172738: device is a tablet
[    50.739] (II) event23 - failed to create input device '/dev/input/event23'.
[    50.739] (EE) libinput: Tablet Monitor Pen 172738: Failed to create a device for /dev/input/event23
[    50.739] (EE) PreInit returned 2 for "Tablet Monitor Pen 172738"
[    50.739] (II) UnloadModule: "libinput"

kalle-ankan, 2019-03-25:
Hi

We got it working now you can close the bug.
It was an typo.
So I can confirm that Huion GT 191 works in Debian 10 (Buster).

joseluis, 2019-03-25:
Good, I'm happy it works

closed 2019-03-25 by joseluis