joseluis/huion-linux-drivers #16: Support for Huin H430P

SandBoxCZ, 2019-03-31:
Hello i would to ask you, if you can add support for Huion H430P :)

joseluis, 2019-04-02:
Hi @SandBoxCZ, I've added `[tablet_h430p]`, please try it and let me know if it works for you.

If it doesn't work for you, I'll ask you to either set `debug_mode = true` or  `current_tablet = [tablet_debug]` again), then go through the suggested *corner touching* routine and copy&paste the output back to me to do some checking on that data.

SandBoxCZ, 2019-04-02:
Hi, it does not work, so im sending you the requested data.

https://pastebin.com/637iX7LL

Hope this log helps fix it :)

joseluis, 2019-04-02:
I've updated the script with better debug info. Please use the latest version (of script and config) and leave everything by default, except for your chose tablet in the config.ini file:

```
current_tablet = [tablet_h430p]
```

thanks!

SandBoxCZ, 2019-04-02:
Here is debug info.

https://pastebin.com/AyJYtaVP

Hope this helps you more :)

joseluis, 2019-04-03:
Hey, thank you for the info. But I still don't know how to make it work. I'll keep working on it, but for now I recommend you to use [latest release of digimend kernel drivers](https://github.com/DIGImend/digimend-kernel-drivers/releases), it added support to your tablet in version V8

joseluis, 2019-04-14:
Hi again, you can also try running this small script, in the colment I link below, which performs a basic test, to rule out a system malfunction, like in the linked issue, similar to yours:

https://github.com/joseluis/huion-linux-drivers/issues/17#issuecomment-480206333

closed 2019-09-20 by joseluis