joseluis/huion-linux-drivers #17: Huion Kamvas Pro 12(GT-116)

Biological-Flask-Mime, 2019-04-02:
In config.ini `current_tablet = [tablet_debug]`

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 53580
    |                          Max Y: 33640
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M171_190122
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name         DEBUG TABLET
        Buttons                   disabled (0)
        Scrollbar                 disabled (0)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
        [Debug mode only]. Input from tablet wont be used, except
        for printing out the information to the console.

        INSTRUCTIONS: briefly touch the four corners of the screen:
                1) Left up 2) Right up 3) Left Down 4) Right Down
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Huion Kamvas driver should now be running

Please slowly and briefly touch the LEFT UP corner of your tablet:
...
data[12] = array('B', [8, 129, 135, 0, 127, 0, 174, 0, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 134, 0, 128, 0, 195, 1, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 133, 0, 126, 0, 153, 2, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 132, 0, 126, 0, 65, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 131, 0, 126, 0, 208, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 130, 0, 126, 0, 80, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 129, 0, 126, 0, 193, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 128, 0, 126, 0, 32, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 127, 0, 126, 0, 111, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 125, 0, 126, 0, 175, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 124, 0, 126, 0, 223, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 124, 0, 125, 0, 255, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 124, 0, 123, 0, 20, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 125, 0, 121, 0, 31, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 127, 0, 118, 0, 29, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 129, 0, 114, 0, 7, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 130, 0, 110, 0, 209, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 132, 0, 106, 0, 118, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 134, 0, 102, 0, 218, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 134, 0, 98, 0, 200, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 134, 0, 98, 0, 117, 2, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 133, 0, 95, 0, 13, 1, 0, 0, 0, 0])
...
Now touch the RIGHT UP corner of your tablet:
data[12] = array('B', [8, 129, 2, 208, 83, 0, 43, 1, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 4, 208, 84, 0, 84, 2, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 5, 208, 84, 0, 75, 3, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 6, 208, 84, 0, 29, 4, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 7, 208, 85, 0, 217, 4, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 8, 208, 86, 0, 141, 5, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 9, 208, 87, 0, 72, 6, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 10, 208, 88, 0, 5, 7, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 11, 208, 88, 0, 186, 7, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 87, 0, 95, 8, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 87, 0, 240, 8, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 88, 0, 103, 9, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 89, 0, 191, 9, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 89, 0, 249, 9, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 88, 0, 28, 10, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 12, 208, 87, 0, 49, 10, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 11, 208, 86, 0, 57, 10, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 10, 208, 86, 0, 60, 10, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 9, 208, 86, 0, 49, 10, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 9, 208, 87, 0, 21, 10, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 9, 208, 88, 0, 225, 9, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 8, 208, 90, 0, 141, 9, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 8, 208, 91, 0, 14, 9, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 7, 208, 92, 0, 82, 8, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 7, 208, 93, 0, 95, 7, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 7, 208, 93, 0, 24, 6, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 5, 208, 92, 0, 117, 4, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 3, 208, 90, 0, 178, 2, 0, 0, 6, 0])
data[12] = array('B', [8, 129, 2, 208, 87, 0, 250, 0, 0, 0, 6, 0])
...
Now touch the LEFT UP corner of your tablet:
data[12] = array('B', [8, 129, 34, 0, 142, 130, 218, 0, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 27, 0, 167, 130, 63, 2, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 19, 0, 190, 130, 53, 3, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 12, 0, 209, 130, 228, 3, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 7, 0, 224, 130, 101, 4, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 3, 0, 235, 130, 188, 4, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 1, 0, 243, 130, 241, 4, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 1, 0, 246, 130, 18, 5, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 2, 0, 244, 130, 38, 5, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 4, 0, 238, 130, 46, 5, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 7, 0, 228, 130, 50, 5, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 13, 0, 217, 130, 28, 5, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 20, 0, 204, 130, 225, 4, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 29, 0, 191, 130, 120, 4, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 38, 0, 177, 130, 205, 3, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 45, 0, 164, 130, 189, 2, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 50, 0, 151, 130, 119, 1, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 54, 0, 138, 130, 44, 0, 0, 0, 12, 0])
...
Now touch the RIGHT DOWN corner of your tablet:
data[12] = array('B', [8, 129, 235, 207, 101, 129, 116, 0, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 237, 207, 101, 129, 74, 1, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 239, 207, 101, 129, 12, 2, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 241, 207, 100, 129, 179, 2, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 243, 207, 100, 129, 66, 3, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 245, 207, 100, 129, 195, 3, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 247, 207, 100, 129, 65, 4, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 248, 207, 100, 129, 199, 4, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 100, 129, 86, 5, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 100, 129, 239, 5, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 101, 129, 142, 6, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 103, 129, 50, 7, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 105, 129, 212, 7, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 108, 129, 105, 8, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 111, 129, 238, 8, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 115, 129, 94, 9, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 119, 129, 186, 9, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 123, 129, 254, 9, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 127, 129, 47, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 249, 207, 130, 129, 82, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 248, 207, 133, 129, 106, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 247, 207, 136, 129, 124, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 246, 207, 139, 129, 136, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 245, 207, 142, 129, 144, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 243, 207, 145, 129, 145, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 241, 207, 148, 129, 125, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 238, 207, 151, 129, 69, 10, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 235, 207, 152, 129, 202, 9, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 231, 207, 152, 129, 250, 8, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 225, 207, 151, 129, 139, 7, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 218, 207, 149, 129, 170, 5, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 212, 207, 144, 129, 156, 3, 0, 0, 16, 252])
data[12] = array('B', [8, 129, 207, 207, 136, 129, 144, 1, 0, 0, 16, 252])
```

joseluis, 2019-04-02:
Hi @Biological-Flask-Mime I've added `[tablet_gt116]` to the config.ini. You can choose that one now and try it.

The X, Y and pressure data points seems to be consistent with the previously known tablets (they communicate through the indexes [2] to [8]). So the tablet should work. This is the summarized translated known data points that looks well:

```
Please slowly and briefly touch the LEFT UP corner of your tablet:
...
{'X': '135', 'Y': '127', 'PRESS': '174'}
{'X': '125', 'Y': '121', 'PRESS': '1567'}
{'X': '133', 'Y': '95', 'PRESS': '269'}
...
Now touch the RIGHT UP corner of your tablet:
{'X': '53250', 'Y': '83', 'PRESS': '299'}
{'X': '53258', 'Y': '86', 'PRESS': '2620'}
{'X': '53250', 'Y': '87', 'PRESS': '250'}
...
Now touch the LEFT DOWN corner of your tablet:
{'X': '34', 'Y': '33422', 'PRESS': '218'}
{'X': '7', 'Y': '33508', 'PRESS': '1330'}
{'X': '54', 'Y': '33418', 'PRESS': '44'}
...
Now touch the RIGHT DOWN corner of your tablet:
{'X': '53227', 'Y': '33125', 'PRESS': '116'}
{'X': '53235', 'Y': '33169', 'PRESS': '2705'}
{'X': '53199', 'Y': '33160', 'PRESS': '400'}
```

Although it also seems to communicate some extra data through [10] & [11]. Do you have any idea what that data could represent?

Could you try your tablet and let me know if everything seems to work well?

Biological-Flask-Mime, 2019-04-02:
Thank you. I've updated the config.ini, however, my pen does not work.

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 53580
    |                          Max Y: 33640
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M171_190122
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name         GT-116
        Buttons                   disabled (4)
        Scrollbar                 disabled (1)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    Enabled (1920x1080)
        Current Monitor Setup     [monitor_2]
        Multi Monitor Setup       disabled
        Calling xrandr            disabled

Huion Kamvas driver should now be running
```
Nothing happens

I think I see what extra data you're talking about but I am not too sure what it is. I think it could be the pen tilt because when i keep the pen straight it's value is 0 and when i tilt it slightly it might go to 16 or something but having it almost horizontal to the tablet it goes to around 216.

This tablet is my only monitor. I'm not sure if that could change anything.

Going back into the debug mode:

```
data[12] = array('B', [8, 129, 135, 105, 138, 72, 250, 11, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 95, 105, 128, 72, 251, 11, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 54, 105, 118, 72, 254, 11, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 11, 105, 108, 72, 4, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 224, 104, 98, 72, 13, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 180, 104, 89, 72, 26, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 137, 104, 81, 72, 42, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 93, 104, 74, 72, 62, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 50, 104, 68, 72, 85, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 7, 104, 64, 72, 109, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 222, 103, 62, 72, 133, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 182, 103, 62, 72, 157, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 144, 103, 64, 72, 181, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 109, 103, 67, 72, 206, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 76, 103, 71, 72, 233, 12, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 45, 103, 75, 72, 7, 13, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 17, 103, 80, 72, 40, 13, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 246, 102, 84, 72, 71, 13, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 221, 102, 88, 72, 102, 13, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 197, 102, 91, 72, 128, 13, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 175, 102, 93, 72, 147, 13, 0, 0, 255, 0])
data[12] = array('B', [8, 129, 155, 102, 94, 72, 160, 13, 0, 0, 255, 0])
data[12] = array('B', [8, 129, 136, 102, 94, 72, 168, 13, 0, 0, 255, 0])
data[12] = array('B', [8, 129, 119, 102, 94, 72, 173, 13, 0, 0, 255, 0])
data[12] = array('B', [8, 129, 104, 102, 94, 72, 177, 13, 0, 0, 254, 0])
data[12] = array('B', [8, 129, 92, 102, 94, 72, 185, 13, 0, 0, 254, 0])
data[12] = array('B', [8, 129, 81, 102, 94, 72, 197, 13, 0, 0, 254, 0])
data[12] = array('B', [8, 129, 72, 102, 94, 72, 217, 13, 0, 0, 254, 0])
data[12] = array('B', [8, 129, 64, 102, 94, 72, 249, 13, 0, 0, 253, 0])
data[12] = array('B', [8, 129, 58, 102, 95, 72, 37, 14, 0, 0, 253, 0])
data[12] = array('B', [8, 129, 53, 102, 98, 72, 89, 14, 0, 0, 253, 0])
data[12] = array('B', [8, 129, 50, 102, 102, 72, 142, 14, 0, 0, 253, 0])
data[12] = array('B', [8, 129, 49, 102, 107, 72, 193, 14, 0, 0, 252, 0])
data[12] = array('B', [8, 129, 52, 102, 114, 72, 237, 14, 0, 0, 252, 0])
data[12] = array('B', [8, 129, 58, 102, 122, 72, 15, 15, 0, 0, 251, 0])
data[12] = array('B', [8, 129, 68, 102, 131, 72, 36, 15, 0, 0, 251, 0])
data[12] = array('B', [8, 129, 80, 102, 140, 72, 47, 15, 0, 0, 250, 0])
data[12] = array('B', [8, 129, 95, 102, 149, 72, 52, 15, 0, 0, 250, 0])
```
Above, you can see the top half there I have the pen pressed down but kept the pen straight, and the bottom half I tilted the ben to the left.

Below, I had the pen pressed straight then pulled it up and pushed it back down but tilted the pen down.

```
data[12] = array('B', [8, 129, 137, 124, 249, 102, 110, 10, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 136, 124, 248, 102, 43, 9, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 136, 124, 247, 102, 206, 7, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 136, 124, 247, 102, 89, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 136, 124, 247, 102, 203, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 136, 124, 247, 102, 29, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 136, 124, 253, 102, 127, 1, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 137, 124, 4, 103, 7, 0, 0, 0, 0, 0])
...
data[12] = array('B', [8, 129, 214, 122, 180, 98, 2, 0, 0, 0, 6, 205])
data[12] = array('B', [8, 129, 211, 122, 179, 98, 3, 0, 0, 0, 6, 205])
data[12] = array('B', [8, 129, 211, 122, 179, 98, 3, 0, 0, 0, 6, 205])
data[12] = array('B', [8, 129, 211, 122, 179, 98, 3, 0, 0, 0, 6, 205])
data[12] = array('B', [8, 129, 211, 122, 179, 98, 1, 0, 0, 0, 6, 204])
```

joseluis, 2019-04-02:
Fantastic, yes that is actually the tilt: [10] & [11] will be the X & Y axis, respectively.

I've added preliminary tilt support but I suspect it wont work right now and will need more refinement...

> This tablet is my only monitor. I'm not sure if that could change anything.

Maybe... I've added a new single monitor mode, try setting `current_monitor_setup   = [monitor_1]`

Ok, so I've also improved the debug mode. Please set your `current_tablet = [tablet_gt116]` and leave the `debug_mode = true` and show me the output. Let's see if we can see what's happening there.

BTW are you using the latest version of uclogic-tools compiled from source?

Biological-Flask-Mime, 2019-04-02:
Thank you for going out of your way for this. I am using [v5 of uclogic-tools](https://github.com/DIGImend/uclogic-tools/releases), which is the latest release.

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 53580
    |                          Max Y: 33640
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M171_190122
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name         GT-116
        Buttons                   disabled (4)
        Scrollbar                 disabled (1)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    Enabled (1920x1080)
        Current Monitor Setup     [monitor_1]
        Multi Monitor Setup       disabled
        Calling xrandr            disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 5.0.5-arch1-1-ARCH (x86_64 )
#1 SMP PREEMPT Wed Mar 27 17:53:10 UTC 2019
./huion-tablet-driver.py:195: DeprecationWarning: Please use InputDevice.path instead of InputDevice.fn
  print("\nDEVICE: {} ({})".format(device.fn, device.phys))

DEVICE: /dev/input/event22 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

TABLET CAPABILITIES:
[0, 1, 3, 21]
[320, 330, 331, 332]
[(0, AbsInfo(value=0, min=0, max=53580, fuzz=0, flat=0, resolution=0)), (1, AbsInfo(value=0, min=0, max=33640, fuzz=0, flat=0, resolution=0)), (24, AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0))]

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        In order to show useful calibration info then
        Briefly touch the four corners of the screen:
        1) Left up 2) Right up 3) Left Down 4) Right Down
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

...
data[12] = array('B', [8, 129, 3, 1, 145, 0, 201, 0, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 251, 0, 164, 0, 76, 2, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 243, 0, 184, 0, 109, 3, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 235, 0, 205, 0, 83, 4, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 227, 0, 226, 0, 9, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 220, 0, 250, 0, 135, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 214, 0, 17, 1, 217, 5, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 209, 0, 42, 1, 14, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 206, 0, 65, 1, 50, 6, 0, 0, 0, 0])
data[12] = array('B', [8, 129, 206, 0, 85, 1, 73, 6, 0, 0, 0, 0])
[... 42 lines cut ...]
data[12] = array('B', [8, 129, 86, 1, 159, 128, 69, 3, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 87, 1, 153, 128, 33, 3, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 87, 1, 147, 128, 200, 2, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 86, 1, 142, 128, 7, 2, 0, 0, 12, 0])
data[12] = array('B', [8, 129, 85, 1, 138, 128, 15, 1, 0, 0, 12, 0])
...
data[12] = array('B', [8, 129, 188, 207, 178, 128, 132, 0, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 182, 207, 187, 128, 62, 2, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 175, 207, 197, 128, 22, 4, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 168, 207, 208, 128, 118, 5, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 162, 207, 218, 128, 113, 6, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 157, 207, 228, 128, 39, 7, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 152, 207, 237, 128, 168, 7, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 146, 207, 245, 128, 2, 8, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 140, 207, 252, 128, 64, 8, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 134, 207, 2, 129, 94, 8, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 127, 207, 7, 129, 89, 8, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 120, 207, 10, 129, 30, 8, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 113, 207, 11, 129, 166, 7, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 105, 207, 11, 129, 203, 6, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 96, 207, 10, 129, 91, 5, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 87, 207, 8, 129, 170, 3, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 80, 207, 5, 129, 228, 1, 0, 0, 11, 0])
data[12] = array('B', [8, 129, 74, 207, 0, 129, 46, 0, 0, 0, 11, 0])
```

Biological-Flask-Mime, 2019-04-03:
I'm just looking around, since I don't understand a lot of this, but, when I ran the driver, somewhere in there it says 
```
DEVICE: /dev/input/event22 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003
```
Sometimes the device changes from /dev/input/event22 to /dev/input/event23

I looked in /dev/input/ and there are multiple events in there but no event22 or event23

joseluis, 2019-04-03:
Interesting. My tablet is the same except the first capabilities list doesn't have a 21.

I've updated the script with even more debug info. I'm still not sure why your cursor is not moving...

One question. Did you disable the buttons and the scrollbar? Do they work well for you?

Biological-Flask-Mime, 2019-04-03:
This is with the updated scripts

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 53580
    |                          Max Y: 33640
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M171_190122
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name         GT-116
        Buttons                   ENABLED (4)
        Scrollbar                 ENABLED (1)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    Enabled (1920x1080)
        Current Monitor Setup     [monitor_1]
        Multi Monitor Setup       disabled
        Calling xrandr            disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 5.0.5-arch1-1-ARCH (x86_64 )
#1 SMP PREEMPT Wed Mar 27 17:53:10 UTC 2019

DEVICE: /dev/input/event23 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=53580, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=33640, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        In order to share useful calibration info, please:
        Briefly touch the four corners of the screen:
        1) Left up 2) Right up 3) Left Down 4) Right Down
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

...
008 129 006 001 078 001 016 001 000 000 000 000 | X:00262 Y:00334 PRES:0272 TILT_X:000 TILT_Y:000
008 129 255 000 089 001 189 001 000 000 000 000 | X:00255 Y:00345 PRES:0445 TILT_X:000 TILT_Y:000
008 129 248 000 098 001 048 002 000 000 000 000 | X:00248 Y:00354 PRES:0560 TILT_X:000 TILT_Y:000
008 129 243 000 104 001 131 002 000 000 000 000 | X:00243 Y:00360 PRES:0643 TILT_X:000 TILT_Y:000
[... 53 lines cut ...]
008 129 173 208 199 129 255 001 000 000 021 000 | X:53421 Y:33223 PRES:0511 TILT_X:021 TILT_Y:000
008 129 173 208 199 129 093 001 000 000 021 000 | X:53421 Y:33223 PRES:0349 TILT_X:021 TILT_Y:000
008 129 173 208 199 129 147 000 000 000 021 000 | X:53421 Y:33223 PRES:0147 TILT_X:021 TILT_Y:000

```
I had the buttons and scrollbar disabled since I never really use them. I enabled them this time. Below, I pressed the 4 buttons down and then used the scrollbar. Every time I use the scrollbar, it crashes the driver.
```
008 224 001 001 001 000 000 000 000 000 012 000 | X:00257 Y:00001 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 000 000 000 000 000 000 012 000 | X:00257 Y:00000 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 002 000 000 000 000 000 012 000 | X:00257 Y:00002 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 000 000 000 000 000 000 012 000 | X:00257 Y:00000 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 004 000 000 000 000 000 012 000 | X:00257 Y:00004 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 000 000 000 000 000 000 012 000 | X:00257 Y:00000 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 008 000 000 000 000 000 012 000 | X:00257 Y:00008 PRES:0000 TILT_X:012 TILT_Y:000
008 224 001 001 000 000 000 000 000 000 012 000 | X:00257 Y:00000 PRES:0000 TILT_X:012 TILT_Y:000
008 240 001 001 000 001 000 000 000 000 012 000 | X:00257 Y:00256 PRES:0000 TILT_X:012 TILT_Y:000
Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 730, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 34, in run
    main_loop()
  File "./huion-tablet-driver.py", line 409, in main_loop
    do_shortcut("scrollbar", MENU[main.current_menu]['scroll_up'])
KeyError: ''
```

joseluis, 2019-04-04:
Thanks, I just fixed the scrollbar bug.

> I'm just looking around, since I don't understand a lot of this

I feel strangely identified with this statement :sweat_smile: :roll_eyes: 

It seems clear to me that the data from your tablet is being read perfectly fine, but the writing part is not getting applied correctly, or it is not getting through to the system. I need to study more how this linux event system works more deeply, because I don't understand yet why it isn't working for you.

My guess is that this piece of code will need to be modified (around lines 109-120):
```
def setup_driver():                                                                                 
    cap_pen = {                                                                                     
        ecodes.EV_KEY: [ecodes.BTN_TOUCH, ecodes.BTN_TOOL_PEN,ecodes.BTN_STYLUS, ecodes.BTN_STYLUS2],
        ecodes.EV_ABS: [
            # value,min,max,fuzz,flat,resolution
            (ecodes.ABS_X,AbsInfo(0,0,main.settings['pen_max_x'],0,0,main.settings['resolution'])),
            (ecodes.ABS_Y,AbsInfo(0,0,main.settings['pen_max_y'],0,0,main.settings['resolution'])),
            (ecodes.ABS_PRESSURE, AbsInfo(0,0,main.settings['pen_max_z'],0,0,0)),
            (ecodes.ABS_TILT_X, AbsInfo(0,0,255,0,0,0)), # TODO: tweak
            (ecodes.ABS_TILT_Y, AbsInfo(0,0,255,0,0,0)),
        ]
    }
    main.vpen = UInput(events=cap_pen, name=main.settings['pen_device_name'], version=0x3)
```

Since the the writes around lines 434-444 doesn't seem to have any mystery.
```
main.vpen.write(ecodes.EV_ABS, ecodes.ABS_X, X)
main.vpen.write(ecodes.EV_ABS, ecodes.ABS_Y, Y)
# ....
```
 I'll keep investigating when I get some time.

For now, could you tell me which version of python and evdev you have? You can use the following commands:

```
python3 --version
dpkg -l | grep python.*evdev
pip3 list | grep evdev
```

thanks!

Biological-Flask-Mime, 2019-04-04:
Python 3.7.3
evdev in /usr/lib/python3.7/site-packages (1.1.2)

I've used a different driver before and had this tablet working for a while but then one day it stopped working. Maybe I have broken stuff on my system.

joseluis, 2019-04-05:
Ok let's check that then.

I've created a very simple script that should work in any system: [test-evdev.py](https://gist.github.com/joseluis/ace22ea89d22970d17519cd10508d1bb)

If you run it, it should print the letter 'a' to the console, move the mouse cursor to the top left corner (100x100px) and open the right button menu.

Biological-Flask-Mime, 2019-04-05:
I tried to run the script but I think I'm getting [this problem](https://stackoverflow.com/questions/30285154/python-error-command-not-found).
`sudo: ./test-evdev.py: command not found`

Last night, I unplugged the usb for this tablet and plugged in my other tablet, Huion h610pro, a tablet that has worked for me just by plugging it in, but now when I use it, it gives the tablet a 2nd cursor and the cursor is invisible sometimes.
I'm not looking to fix that one, I'm just testing things, and that tablet doesn't work properly anymore either.

I'm going to install the same distro on a new storage drive later and see if a clean install gives anything different. There are more things messed up with my system and I have a feeling that it's my fault for this not working. lol

joseluis, 2019-04-05:
> I tried to run the script but I think I'm getting this problem.
sudo: ./test-evdev.py: command not found

Ok. You need to save that file with the name test-evdev.py, give it execution permissions and be in the same folder when you run it... Like this:

```
wget https://gist.githubusercontent.com/joseluis/ace22ea89d22970d17519cd10508d1bb/raw/c154c9eda7901c2599314083c07ba6d5dbe43a73/test-evdev.py -O test-evdev.py
chmod +x test-evdev.py
sudo ./test-evdev.py
```

Yes. If you can try this driver in a new install, or even a Ubuntu or Mint LiveCD or something like that that should also give us more information.

Biological-Flask-Mime, 2019-04-05:
> Ok. You need to save that file with the name test-evdev.py, give it execution permissions and be in the same folder when you run it... Like this:

Right... Sorry about that lol. 

I ran it this time and nothing happened. No error but nothing else either.

joseluis, 2019-04-07:
Thats unusual. Could you please tell me the output of these commands?:

`ll /dev/uinput`

`cat /etc/X11/xorg.conf.d/evdev-tablet.conf`

Biological-Flask-Mime, 2019-04-07:
the `ll` command was not found but, if `ls -l` does the same thing, here are my outputs:
```

[rundel@Deusus ~]$ ls -l /dev/uinput
crw-rw----+ 1 root root 10, 223 Apr  7 10:26 /dev/uinput
[rundel@Deusus ~]$ cat /etc/X11/xorg.conf.d/evdev-tablet.conf
Section "InputClass"
        Identifier "evdev tablet catchall"
        MatchIsTablet "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection[rundel@Deusus ~]$ 
```

joseluis, 2019-04-07:
That looks well so I'm out of ideas for now. This is system-related...

Biological-Flask-Mime, 2019-04-13:
I just wanted to update. I reinstalled my operating system on a new SSD and followed all the steps. The tablet works, and does not have the diagonal line wobble. Thank you very much @joseluis

joseluis, 2019-04-14:
That's wonderful. Thanks.

closed 2019-04-14 by joseluis