joseluis/huion-linux-drivers #18: GT-220 v2 - Cursor does not respond to input

MatisHudon, 2019-04-18:
Hi!

Thanks for all the work. I am on ubuntu 16.04 and I have a Huion GT-220 v2. I followed all the instructions carefully. When I launch the driver all seem to work except that the pen does not make the mouse move at all. The click seem to work though. I ran the debug script and here is my output: 
`Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 37540
    |                          Max Y: 21120
    |                   Max pressure: 8191
    |                     Resolution: 2000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 95352
    |                          Max Y: 53645
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M165_170518
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name         GT-220 v2
	Buttons                   disabled (0)
	Scrollbar                 disabled (0)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    Enabled (1920x1080)
	Current Monitor Setup     [monitor_3]
	Multi Monitor Setup       ENABLED
	Calling xrandr            ENABLED

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.16.3-041603-generic (x86_64 x86_64)
#201804190730 SMP Thu Apr 19 07:32:02 UTC 2018

DEVICE: /dev/input/event31 ()
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=95352, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=53645, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=8191, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=0, fuzz=255, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=0, fuzz=255, flat=0, resolution=0))]
VPEN:
name "Tablet Monitor Pen 135001", bus "BUS_USB", vendor "0001", product "0001", version "0003"
event types: EV_SYN EV_KEY EV_ABS

Setting up multiple monitors. . . 
Running xrandr. . . 
» xrandr --output DVI-D-0 --mode 1920x1080 --pos 0x0 --rotate normal --output DP-6 --mode 3840x2160 --pos 1920x0 --rotate normal --output DP-0 --mode 3840x2160 --pos 5760x0 --rotate left
Running xinput. . . 
» xinput set-prop "Tablet Monitor Pen 135001" --type=float "Coordinate Transformation Matrix" 0.2 0 0.0 0 1.0 0.0 0 0 1
Mapped tablet area to "1920x1080 + 0x0"

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

08 81 d9 14 ab 79 66 01 01 00 00 00 | X:70873 Y:31147 PRES:0358 TILT_X:000 TILT_Y:000
08 81 80 14 9a 79 65 01 01 00 00 00 | X:70784 Y:31130 PRES:0357 TILT_X:000 TILT_Y:000
08 81 50 14 9e 79 67 01 01 00 00 00 | X:70736 Y:31134 PRES:0359 TILT_X:000 TILT_Y:000
08 81 1e 14 a1 79 65 01 01 00 00 00 | X:70686 Y:31137 PRES:0357 TILT_X:000 TILT_Y:000
08 81 ba 13 a5 79 65 01 01 00 00 00 | X:70586 Y:31141 PRES:0357 TILT_X:000 TILT_Y:000
08 81 4c 13 a2 79 67 01 01 00 00 00 | X:70476 Y:31138 PRES:0359 TILT_X:000 TILT_Y:000
08 81 d6 12 98 79 67 01 01 00 00 00 | X:70358 Y:31128 PRES:0359 TILT_X:000 TILT_Y:000
08 81 57 12 89 79 67 01 01 00 00 00 | X:70231 Y:31113 PRES:0359 TILT_X:000 TILT_Y:000
08 81 ce 11 74 79 67 01 01 00 00 00 | X:70094 Y:31092 PRES:0359 TILT_X:000 TILT_Y:000
08 81 46 11 59 79 69 01 01 00 00 00 | X:69958 Y:31065 PRES:0361 TILT_X:000 TILT_Y:000
08 81 bf 10 3b 79 6a 01 01 00 00 00 | X:69823 Y:31035 PRES:0362 TILT_X:000 TILT_Y:000
08 81 3e 10 19 79 6c 01 01 00 00 00 | X:69694 Y:31001 PRES:0364 TILT_X:000 TILT_Y:000
08 81 c7 0f f4 78 6d 01 01 00 00 00 | X:69575 Y:30964 PRES:0365 TILT_X:000 TILT_Y:000
08 81 74 0f dd 78 6f 01 01 00 00 00 | X:69492 Y:30941 PRES:0367 TILT_X:000 TILT_Y:000
`

Also I need to mention that I had to create the xorg.conf.d folder in /etc/X11/.
My configuration is with 3 screens although I tried only with the Huion as main screen and had the same result.Here is my xrandr output:

Thanks for your help!

joseluis, 2019-04-18:
Hello @MatisHudon 

Let's first see if your system succesfully writes mouse events.

I created a very simple script that should work in any system: [test-evdev.py](https://gist.github.com/joseluis/ace22ea89d22970d17519cd10508d1bb) (I'm thinking on integrating it in the main script, but for now...)

If you run it, it should print the letter 'a' to the console, move the mouse cursor to the top left corner (100x100px) and open the right button menu.

You can follow the following instructions:
```sh
wget https://gist.githubusercontent.com/joseluis/ace22ea89d22970d17519cd10508d1bb/raw/c154c9eda7901c2599314083c07ba6d5dbe43a73/test-evdev.py -O test-evdev.py
chmod +x test-evdev.py
sudo ./test-evdev.py
```

If that doesn't work, the problem wont be in the script, but in your system.

MatisHudon, 2019-04-18:
Hey, 

Thanks for your help. I ran your test script and:
"a" was printed and the right button menu was opened however the mouse wasn't move to the top left corner which seems to be coherent with the problem described above.

Somehow only mouse moves are not successfully written.. 

Thanks again,

joseluis, 2019-04-18:
Good to know. Then my recommendation for you is to try it script in a new OS installation. Because I wouldn't know how to debug what's wrong there... Feel free to update/reopen if you get some update.

Best of luck

closed 2019-04-18 by joseluis

MatisHudon, 2019-04-23:
Hello, 

I managed to solve the issue, now the driver is working as expected I guess. So I did not re-install the OS. The issue was that I already had a older version of evdev installed for python3 and somehow the pip3 package was not downloading the last version of evdev. I had to:
``
sudo pip3 install --no-cache-dir --upgrade evdev
``
to force the upgrade of evdev after that all worked fine even the pressure sensitivity in Krita. Thanks for the help and the good work.
(I also updated/upgraded all my packages)

joseluis, 2019-04-23:
Very good to know! Thanks for the update and for the detailed explanation, it will be for sure of help for others.