joseluis/huion-linux-drivers #22: Cursor moves to lower right corner when pen comes close to screen.

EricGebhart, 2019-07-13:
The driver seems to be working but the mouse moves to the lower right corner any time the pen
comes close to the screen.  it seems to be registering all of the movements of the pen just fine.

The scrollbar causes the mouse pointer to appear to the left and to move up and down.
after letting go of the scroll bar the pointer jumps to the top of the screen.

Putting the pen close to the screen moves the pointer to the bottom right corner.

This is a fresh install of Arch Linux from last week.  07/07/2019. 
I had installed the wacom drivers but have uninstalled them since.

Kernel is 5.1

i made the entry in the xorg. conf  as stated in the install guide.
I changed the name to tablet_gt133  and that's the only change i made to
the config.

Here is the startup from the driver log.

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 58752
    |                          Max Y: 33048
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M182_181206
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name         GT-133
	Buttons                   ENABLED (5)
	Scrollbar                 ENABLED (1)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    Enabled (1920x1080)
	Current Monitor Setup     [monitor_2]
	Multi Monitor Setup       disabled
	Calling xrandr            disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 5.1.15-arch1-1-ARCH (x86_64 )
#1 SMP PREEMPT Tue Jun 25 04:49:39 UTC 2019

DEVICE: /dev/input/event20 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=58752, fuzz=0, flat=0, resolution=5080)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=33048, fuzz=0, flat=0, resolution=5080)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]
VPEN:
name "Tablet Monitor Pen 143242", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

```
grep evdev of  the xorg.log looks ilke this.

```
[ 11825.864] (**) Power Button: Applying InputClass "evdev keyboard catchall"
[ 11825.868] (**) Video Bus: Applying InputClass "evdev keyboard catchall"
[ 11825.869] (**) Power Button: Applying InputClass "evdev keyboard catchall"
[ 11825.871] (**) Sleep Button: Applying InputClass "evdev keyboard catchall"
[ 11825.873] (**) ErgoDox EZ ErgoDox EZ: Applying InputClass "evdev keyboard catchall"
[ 11825.877] (**) ErgoDox EZ ErgoDox EZ Mouse: Applying InputClass "evdev pointer catchall"
[ 11825.880] (**) ErgoDox EZ ErgoDox EZ System Control: Applying InputClass "evdev keyboard catchall"
[ 11825.883] (**) ErgoDox EZ ErgoDox EZ Consumer Control: Applying InputClass "evdev keyboard catchall"
[ 11825.887] (**) ErgoDox EZ ErgoDox EZ Keyboard: Applying InputClass "evdev keyboard catchall"
[ 11825.893] (**) ErgoDox EZ ErgoDox EZ Consumer Control: Applying InputClass "evdev keyboard catchall"
[ 11873.576] (**) Tablet Monitor Pen 134947: Applying InputClass "evdev tablet catchall"
[ 11873.576] (**) Tablet Monitor Pen 134947: Applying InputClass "evdev tablet catchall"
[ 11873.576] (II) LoadModule: "evdev"
[ 11873.576] (II) Loading /usr/lib/xorg/modules/input/evdev_drv.so
[ 11873.576] (II) Module evdev: vendor="X.Org Foundation"
[ 11873.576] (II) Using input driver 'evdev' for 'Tablet Monitor Pen 134947'
[ 11873.577] (**) evdev: Tablet Monitor Pen 134947: Device: "/dev/input/event10"
[ 11873.577] (--) evdev: Tablet Monitor Pen 134947: Vendor 0x1 Product 0x1
[ 11873.577] (--) evdev: Tablet Monitor Pen 134947: Found absolute axes
[ 11873.577] (--) evdev: Tablet Monitor Pen 134947: Found x and y absolute axes
[ 11873.577] (--) evdev: Tablet Monitor Pen 134947: Found absolute tablet.
[ 11873.577] (II) evdev: Tablet Monitor Pen 134947: Configuring as tablet
[ 11873.577] (**) evdev: Tablet Monitor Pen 134947: YAxisMapping: buttons 4 and 5
[ 11873.577] (**) evdev: Tablet Monitor Pen 134947: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[ 11873.577] (II) evdev: Tablet Monitor Pen 134947: initialized for absolute axes
```

joseluis, 2019-07-18:
Hello @EricGebhart I believe this can happen if the screens are not properly configured in in the `config.ini` file.

How many monitors do you have connected, including the tablet screen, and what are their resolutions?

E.g. If you have 2 monitors, you have to set `enable_multi_monitor` to `true` in line 30. And then write the correct widths and heights in the `[monitor_2]` section at lines 285 and following.

EricGebhart, 2019-07-18:
The huion is my only screen.

joseluis, 2019-07-19:
Ok. Try changing these config rules and let me know if it works:

```
enable_multi_monitor    = true
current_monitor_setup   = [monitor_1]
```

EricGebhart, 2019-07-19:
That didn't go so well... 

It looked ok up until then.

modprobe: FATAL: Module uinput not found in directory /lib/modules/5.1.15-arch1-1-ARCH
Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 744, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 34, in run
    multi_monitor()
  File "./huion-tablet-driver.py", line 233, in multi_monitor
    C1=(main.settings['tablet_offset_x'] / main.settings['total_screen_width'])
KeyError: 'tablet_offset_x'

joseluis, 2019-07-22:
Ok that is a bug with an easy fix. You need to add the following lines to `config.ini` inside the  `[monitor_1]` section. For example just after the line 261:

```
tablet_offset_x      = 0
tablet_offset_y      = 0
```

EricGebhart, 2019-07-22:
That worked better here is the log.

[huion.log](https://github.com/joseluis/huion-linux-drivers/files/3418359/huion.log)

But it still died in the end with missing uinput.

modprobe: FATAL: Module uinput not found in directory /lib/modules/5.1.15-arch1-1-ARCH

EricGebhart, 2019-07-22:
I  rechecked that I installed everything and that my X tablet config was correct.  It is as the readme says for it to be.

joseluis, 2019-07-25:
Ok. This error is dependant on your particular system. It means your kernel doesn't have the required module.

Either you compiled the kernel yourself and didn't enable that module, or you are running a different kernel version after an update than the one the modules are installed for, or you need to rebuild the kernel...

The solution may be as easy as to reboot your system after an update, in case you haven't done so in weeks....

If that doesn't work you should [google for alternative solutions you can try](https://www.google.com/search?q=%22modprobe%3A+FATAL%3A%22+Module+%22uinput%22+%22not+found+in+directory%22+%22arch%22).

EricGebhart, 2019-08-03:
Ok. I thought I rebooted after my last update but apparently not.  
It's working now.

It is a little strange that I have to change it to a multi-monitor setup with one monitor for it work.
Either the code or the instructions should probably be updated.
Thanks for your help

closed 2019-08-06 by joseluis