joseluis/huion-linux-drivers #25: Wayland support

DavidBM, 2019-08-21:
Hi! 

I would love to use this in wayland. I'm down to help and contribute. Did you have any though on this?

I guess that xrand won't be an option in wayland.

joseluis, 2019-08-28:
Hey there,

I'm open collaborations, but I haven't thought about Wayland compatibility before. Although the xrand part is not an essential part of the driver, but merely a convenience, there maybe other things that can be improved too. Let me know if you have something.

t0cc, 2019-12-05:
@DavidBM For Wayland you have already very good support build in most desktop environments. Plasma and Gnome have nice GUIs you can use. Unfortunately, not all distros install them by default. Just look around a bit.

DavidBM, 2019-12-06:
Hi! 
I created my own driver for that. It was a good learning experience. In any case, in ubuntu 19.10 there is native support, non external driver is needed at all.

closed 2020-04-28 by joseluis