joseluis/huion-linux-drivers #27: No need to build uclogic, it's available as package.

triantares, 2019-09-12:
That was the good news, the bad news is:
Debian10 Buster "uclogic-tools.deb" installs "uclogic-probe" and "uclogic-decode" in /usr/bin/, so not in /us/local/bin/ as set in config.ini.
It will need to be found now.

joseluis, 2019-09-20:
I wanted to keep that part simple, so if your uclogic is in another place it's easy to change that line in the config file. Do you think that works for you?

joseluis, 2019-09-20:
I have added this issue to the new [troubleshooting section](https://github.com/joseluis/huion-linux-drivers#troubleshooting-) on the readme.

triantares, 2019-09-20:
<quote>. Do you think that works for you?</quote>
Obviously it worked for me but might not for a lot of others.
Maybe adding a comment there in the config file to save an unsuspecting soul some searching on the net, might work?

closed 2019-09-20 by triantares