joseluis/huion-linux-drivers #29: Multi Monitor

fukurouchu, 2019-09-21:
Opening this issue due to the drivers not working with multi monitor anymore, they used to before but now I just can't.

The drivers still work if I only use 1 monitor.

Otherwise if enable_multi_monitor    = true  I get the following:

> [chu@nest huion-linux-drivers]$ sudo python3 huion-tablet-driver.py
> Finding USB device. . . Done!
> Reading configuration. . . Done!
> Preparing driver. . . Done!
> --------------------------------------------------------------------------------
>  c8 |        Params block #2: ????????
>     |                          Max X: 86970
>     |                          Max Y: 47752
>     |                   Max pressure: 8191
>     |                     Resolution: 5080
>     |
>  c9 |       Firmware version: HUION_M168_171103
>  ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
> --------------------------------------------------------------------------------
> Setting up driver. . . Done!
> 	Tablet model name         GT-191
> 	Buttons                   disabled (0)
> 	Scrollbar                 disabled (0)
> 	Notifications:            ENABLED
> 		for buttons       ENABLED
> 		for scrollbar     disabled
> 	Screen                    Enabled (1920x1080)
> 	Current Monitor Setup     [monitor_2]
> 	Multi Monitor Setup       ENABLED
> 	Calling xrandr            disabled
> 
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 			< DEBUG MODE ENABLED >
> Enabled by default. You can disable it by setting debug_mode = false
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 
> SYSTEM: Linux 5.2.14-arch2-1-ARCH (x86_64 )
> #1 SMP PREEMPT Thu Sep 12 10:42:38 UTC 2019
> 
> DEVICE: /dev/input/event13 (py-evdev-uinput)
> bus: 0003, vendor 0001, product 0001, version 0003
> 
>       ENDPOINT 0x81: Interrupt IN ==========================
>        bLength          :    0x7 (7 bytes)
>        bDescriptorType  :    0x5 Endpoint
>        bEndpointAddress :   0x81 IN
>        bmAttributes     :    0x3 Interrupt
>        wMaxPacketSize   :   0x40 (64 bytes)
>        bInterval        :    0x2
> 
> TABLET CAPABILITIES:
> [('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
> [(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
> [(('ABS_X', 0), AbsInfo(value=0, min=0, max=86970, fuzz=0, flat=0, resolution=5080)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=47752, fuzz=0, flat=0, resolution=5080)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]
> VPEN:
> name "Tablet Monitor Pen 004707", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
> event types: EV_SYN EV_KEY EV_ABS
> 
> Setting up multiple monitors. . .
> Running xinput. . .
> » xinput set-prop "Tablet Monitor Pen 004707" --type=float "Coordinate Transformation Matrix" 0.6521739130434783 0 0.34782608695652173 0 1.0 0.0 0 0 1
> X Error of failed request:  BadMatch (invalid parameter attributes)
>   Major opcode of failed request:  131 (XInputExtension)
>   Minor opcode of failed request:  57 ()
>   Serial number of failed request:  20
>   Current serial number in output stream:  21
> ERROR running the following comand:
> 	xinput set-prop "Tablet Monitor Pen 004707" --type=float "Coordinate Transformation Matrix" 0.6521739130434783 0 0.34782608695652173 0 1.0 0.0 0 0 1
> RETURN CODE: 1

Queatz, 2019-09-25:
I had to log out and in again to get past `xinput set-prop "Tablet Monitor Pen 004707"`

Might also have something to do with this:https://github.com/joseluis/huion-linux-drivers#xorg-extra-code-

joseluis, 2019-10-03:
Hey @fukurouchu,

Have you added the extra code for xOrg that @Queatz linked to before? You should do that, then reboot and maybe try the last version of the script + config.

fukurouchu, 2019-10-07:
@joseluis  @Queatz  Yeh that was it, I was missing the file, because I formated and forgot I didn't have it anymore, thank you for your help and sorry for the bother.

closed 2019-10-07 by fukurouchu