joseluis/huion-linux-drivers #31: Driver should not require sudo

tnbp, 2019-09-21:
The driver should not require being run with `sudo`. Instead, README should include instructions to set up `udev` rules (e.g. creating `/etc/udev/rules.d/99-huion-tablet.rules` that sets mode = 0664 to a specific group, e.g. `plugdev` or `dialout`).

I would create a pull request, but I'm not very familiar with the udev syntax. My rule:

`SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0664", GROUP="dialout"`

gives write access to USB devices period. I think something like
`SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", MODE="0664", GROUP="dialout"`
`SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006e", MODE="0664", GROUP="dialout"`
might work.

ishbosamiya, 2019-09-26:
Damn, this is much better than what I am currently doing, I should try this out soon.

Desulate, 2020-06-08:
This issue is more than just an "enhancement", It is a security issue and it should be treated as such. At least that is how I view it.

Even if users trust the developer here, there is always the possibility that as this driver is adopted by intermediate users that it could be intercepted or modified to be malicious if 3rd parties start providing it downstream. 

A new user with lower tech literacy, naive, and not knowing the difference may then refer to your install instructions - if they even go out of there way take even that step -. Only to see that "sudo" was officially used upstream in the instructions and then just assume "it must be safe" since its not an "oddity" between your instructions and the 3rd partys. Then proceed to literally just cut and paste the install instructions into a terminal without a second thought.

Additionally requiring the use of sudo when its not absolutely a requisite is going to reduce the likely-hood this package is going to be supported by other projects or adopted by intermediate users, distro user repos, and those with workstation usecases.

With all that said, thank you for creating this driver. It is very much appreciated that someone is properly working to support Huion products and are not just slapping them together along with Wacom drivers. Particularly since Huion on average is really Wacom's best competitor in the space.