joseluis/huion-linux-drivers #33: Kamvas Pro 16: Tilt does not work in applications

floopfloopfloopfloopfloop, 2019-09-21:
# 2: Tilt is doesn't work in applications

Driver debug reports tilt functioning
```
...
08 00 d1 06 af 05 00 00 00 00 10 fd | X:01745 Y:01455 PRES:0000 TILT_X:016 TILT_Y:-253
...
08 00 cb bc 9e 4c 00 00 00 00 2b 0b | X:48331 Y:19614 PRES:0000 TILT_X:043 TILT_Y:-11
...
08 00 48 b5 c5 5d 00 00 00 00 2d 1d | X:46408 Y:24005 PRES:0000 TILT_X:045 TILT_Y:-29
...
08 00 96 ae 79 3f 00 00 00 00 08 db | X:44694 Y:16249 PRES:0000 TILT_X:008 TILT_Y:-219
...
08 00 99 b4 5c 3f 00 00 00 00 04 da | X:46233 Y:16220 PRES:0000 TILT_X:004 TILT_Y:-218
...
08 00 83 bc 55 5c 00 00 00 00 30 05 | X:48259 Y:23637 PRES:0000 TILT_X:048 TILT_Y:-05
...
08 00 40 ce 18 31 00 00 00 00 f9 d0 | X:52800 Y:12568 PRES:0000 TILT_X:249 TILT_Y:-208
...
08 00 f1 c6 7f 43 00 00 00 00 2d 01 | X:50929 Y:17279 PRES:0000 TILT_X:045 TILT_Y:-01
...
08 00 12 cb b2 3f 00 00 00 00 22 f3 | X:51986 Y:16306 PRES:0000 TILT_X:034 TILT_Y:-243
...
08 00 22 cf 22 44 00 00 00 00 26 f5 | X:53026 Y:17442 PRES:0000 TILT_X:038 TILT_Y:-245
...
08 00 d7 de be 3f 00 00 00 00 f5 dd | X:57047 Y:16318 PRES:0000 TILT_X:245 TILT_Y:-221
...
08 00 60 ce 1f 64 00 00 00 00 0f 10 | X:52832 Y:25631 PRES:0000 TILT_X:015 TILT_Y:-16
...
08 00 80 cb 65 5b 00 00 00 00 2e 0d | X:52096 Y:23397 PRES:0000 TILT_X:046 TILT_Y:-13
...
08 00 f7 cb 92 52 00 00 00 00 32 02 | X:52215 Y:21138 PRES:0000 TILT_X:050 TILT_Y:-02
...
08 00 20 e0 25 44 00 00 00 00 f4 e0 | X:57376 Y:17445 PRES:0000 TILT_X:244 TILT_Y:-224

```

But in applications like krita and gimp, tilt brush functionality doesn't work.

```
5.0.0-29-generic #31~18.04.1-Ubuntu SMP Thu Sep 12 18:29:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

Seems to happens with all configs, but here's the current wip: https://gist.github.com/floopfloopfloopfloopfloop/3bf329e1bdfedcbc4e16c2c8d1d0b907

joseluis, 2019-09-25:
The tilt part is just a [non-working placeholder](https://github.com/joseluis/huion-linux-drivers/blob/49b441f0c5851011bdae13e4b428deb0b3b5ebdc/huion-tablet-driver.py#L443), it doesn't do anything yet. Since I don't have a tablet with tilt support I couldn't investigate it further.

t0cc, 2019-12-05:
Is there any chance to get tilt working this year? Maybe you get a Huion tablet with tilt support. I would have lent you one, unfortunatley Spain is too far away for me. Or can we help you in another way?

Pixeliz3d, 2020-03-10:
I am working on this issue. I hope the fix will be ​​ready soon.

PR #66 (2020-04-28): Add tilt support
This PR fixes #33

Works for me with Kamvas Pro 16
closed 2020-04-28 by joseluis