joseluis/huion-linux-drivers #34: Kamvas Pro 16: Enabling on-tablet buttons hangs the driver

floopfloopfloopfloopfloop, 2019-09-21:
wip config: https://gist.github.com/floopfloopfloopfloopfloop/3bf329e1bdfedcbc4e16c2c8d1d0b907

output:

```
$ sudo ./huion-tablet-driver.py 
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
 c8 |        Params block #2: ????????
    |                          Max X: 68840
    |                          Max Y: 38720
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M183_190806
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name         KAMVAS PRO 16
        Buttons                   ENABLED (7)
        Scrollbar                 ENABLED (1)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     ENABLED
        Screen                    Enabled (1920x1080)
        Current Monitor Setup     [monitor_2]
        Multi Monitor Setup       ENABLED
        Calling xrandr            ENABLED

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 5.0.0-29-generic (x86_64 x86_64)
#31~18.04.1-Ubuntu SMP Thu Sep 12 18:29:21 UTC 2019

DEVICE: /dev/input/event15 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=68840, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=38720, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]
VPEN:
name "Tablet Monitor Pen 194525", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS
Calibrating. . . Done!

Setting up multiple monitors. . . 
Running xrandr. . . 
» xrandr --output HDMI-0 --mode 1920x1080 --pos 0x0 --rotate normal --output DP-4 --mode 1920x1080 --pos 1920x0 --rotate normal
Running xinput. . . 
» xinput set-prop "Tablet Monitor Pen 194525" --type=float "Coordinate Transformation Matrix" 0.5 0 0.0 0 1.0 0.0 0 0 1
Mapped tablet area to "1920x1080 + 0x0"

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

[menu_simple_7b]
button 0 = key Tab           # hide interface
button 1 = key ctrl+c        # copy
button 2 = key ctrl+v        # paste
button 3 = key ctrl+z        # undo
button 4 = key ctrl+y        # redo (gimp)
button 5 = key ctrl+shift+z  # redo (krita)
button 6 = key e             # toggle eraser (krita)

```

tablet is non-responsive to all input from this point forward

commenting out  `start_menu = [...]` section makes tablet responsive to pen input + pen buttons.

floopfloopfloopfloopfloop, 2019-09-21:
dazed and confused note: I've also noticed situations where in a text editor, the pen input location sometimes becomes misaligned with the text cursor location. Seems to be related to pressing an on-tablet button beforehand, but amidst the bevy of config fiddling I can't recall under what situations this occurs.

joseluis, 2020-04-28:
Hi. Please try the latest version.

floopfloopfloopfloopfloop, 2020-05-10:
```
$ sudo ./huion-tablet-driver.py 
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
 c8 |        Params block #2: ????????
    |                          Max X: 68840
    |                          Max Y: 38720
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M183_190806
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
        Tablet model name         KAMVAS PRO 16
        Buttons                   ENABLED (7)
        Scrollbar                 ENABLED (1)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    Enabled (1920x1080)
        Current Monitor Setup     [monitor_1]
        Multi Monitor Setup       disabled
        Calling xrandr            disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 5.3.0-51-generic (x86_64 x86_64)
#44~18.04.2-Ubuntu SMP Thu Apr 23 14:27:18 UTC 2020

DEVICE: /dev/input/event4 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=68840, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=38720, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 002113", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
⎜   ↳ Tablet Monitor Pen 002113                 id=8    [slave  pointer  (2)]

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

[menu_4b_simple]
button 0 = key Tab           # hide interface
button 1 = key r             # rect select (gimp) & pick layer (krita)
button 2 = key ctrl+z        # undo
button 3 = key ctrl+y        # redo (gimp)
button 4 = 
button 5 = 
button 6 = 
^C
```

nope, still hangs!

joseluis, 2020-05-11:
HI I just want to say that I'm recreating the software using Rust, it will be more reliable and give less problems that right now are a nightmare to fix because of things related to python packaging and distribution... My intention is to publish a beta this month with similar functionality as the python version has now.

floopfloopfloopfloopfloop, 2020-05-11:
That's funny, I actually partially rewrote your driver in Rust a few months ago as a weekend project! 

I wonder though, is there any advantage to a kernelspace driver? [input-wacom](https://github.com/linuxwacom/input-wacom) seems like a good reference.

joseluis, 2020-05-11:
No kidding. That's so fun.

Well, despite being no expert in drivers at all, I think userspace seems to be enough for the purpose or reading the tablet input, and I suspect the project would become more complex going to kernel-space, and probably less portable, in case that's ever a concern.

Going from python to rust would address any speed issues, nor that anybody complained about that. 

As I see it the needed changes for this project should be relative to eliminate the bugs, make it easy to use, and possible to customize via config files. I'd also like to make it a daemon and control it with a simpe UI.

BTW just for completion I found a couple of examples of kernelspace drivers in rust. Although I don't specially want to dive myself now into the kernel world if I can help it:
* https://github.com/lizhuohua/linux-kernel-module-rust
* https://github.com/tsgates/rust.ko

floopfloopfloopfloopfloop, 2020-05-12:
I have noticed some odd command buffering issues with the current python driver, as the cursor will occasionally lag behind the pen. Not familiar enough to know if it's some python-related internal allocations or a trait of the userspace-to-kernel interactions.

lynxhug, 2020-06-20:
Had similar problem. Try to disable all notifications. For some reason even with installed libnotify it doesn't work at all. There was no error output for me (kubuntu 20.04).