joseluis/huion-linux-drivers #35: Automatic driver loading/unloading

yayfortrees, 2019-09-30:
I was able to setup automatic driver loading/unloading with udev and systemd. It required some minor changes to the code (exit on disconnect), so I'm opening this issue for those changes and adding details on how to setup udev/systemd in case anyone else wants to.

You'll need a systemd service file for the driver:

/etc/systemd/system/huion-driver.service

```
[Unit]
Description=Huion Driver

[Service]
Type=simple
Environment=DISPLAY=:1
Environment=XAUTHORITY=/run/user/1000/gdm/Xauthority
WorkingDirectory=/opt/huion-linux-drivers/
ExecStart=/usr/bin/python3 huion-tablet-driver.py

```
Replace the DISPLAY and XAUTHORITY variables with ones for your system. Set the WorkingDirectory to the location with the driver files.

Next you'll need a udev rule to start the service:

/etc/udev/rules.d/99-huion.rules
```
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idProduct}=="006e", ATTRS{idVendor}=="256c", RUN+="/usr/bin/systemctl --no-block start huion-driver.service"
```
Change the product and vendor IDs to match your device. They can be found with lsusb.

I initially tried having udev start and stop the service on removal, but the driver triggers a removal of the usb device during initialization, udev would then stop the service before it got rolling. I modified the driver code to exit on USB disconnect so I wouldn't have to have udev do it, and it works great.

I'll attach a pull request for the changes, if I can figure this out. ;)

Thanks, the driver is working great!

triantares, 2019-09-30:
This worked exceptionally well and indeed solves the CPU hogging problem I committed.
Thank you very much.
BTW.
I'm wondering why you are using DISPLAY = :1 , where most systems would be using :0 ?

yayfortrees, 2019-09-30:
Glad it worked for you!

On that system gdm stays running on DISPLAY=:0 for additional logins to different VTs. It was configured that way automatically by the Fedora 30 install.

triantares, 2019-10-02:
In that case wouldn't it be better to put "DISPLAY = :0" in your systemd service example, as that would mirror most standard situations ?
It would make it a lot easier for copy/pasters who don't know what display X is on or even what it means. :wine_glass:

yayfortrees, 2019-10-03:
Well, I had hoped to present an example that would capture most installations, but it seems they're split on this. A default Ubuntu install has the first user session on :0 because they have wayland enabled in gdm. If you disable wayland your user session will end up on :1. Fedora and CentOS have wayland disabled in gdm, so the first user session ends up on :1.

I'm not sure why Ubuntu ships with wayland enabled in gdm, but not the user session. Seems like they should choose one or the other.

TJCanterbury, 2020-09-28:
SO I tried this but for some reason it's not working for me. If I do it whilst the tablet is plugged in it wont work and if I unplug and replug the usb I instead start with partial support (no buttons or tilt) from I believe some other driver I've got going on (but don't know which one) instead of the joseluis one, I think this might be interfering? I'm quite new to programming so not sure how best to proceed to solve this. I've got the joseluis driver running perfectly when I do it by sudo.

 /etc/systemd/system/huion-driver.service    
`[Unit]
Description=Huion Tablet Driver

[Service]
Type=simple
Environment=DISPLAY=:1
Environment=XAUTHORITY=/run/user/1000/gdm/Xauthority
WorkingDirectory=/home/tristan/huion-linux-drivers/
ExecStart=/usr/bin/python3 huion-tablet-driver.py
`
 /etc/udev/rules.d/99-huion.rules

`ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idProduct}=="006e", ATTRS{idVendor}=="256c", RUN+="/usr/bin/systemctl --no-block start hu$
`

Mimerr, 2020-11-28:
I tried this as well. First I was trying it before I found this and I didn't have the environment settings in the service or the udev rules. I added those and nothing changed. I get the same error in my systemctl status: **ImportError: sys.meta_path is None, Python is likely shutting down**

I was kind of thinking maybe the environment and udev rules are to get rid of this because I'm guessing it adds the ability to access the usb? I'm not exactly an expert in this area so could be totally wrong.

I am using KDE, so I do not have /run/user/1000/**gdm**/Xauthority since that seems to be a gnome thing. Is there a KDE equivalent?