joseluis/huion-linux-drivers #37: Kamvas 16 pro support for config

sebRomeo, 2019-09-30:
Hi, first great thanks for such a piece of genious :heart_eyes: 

```
[tablet_kamvas_pro16]
model_name    = KAMVAS PRO 16
screen        = true
screen_width  = 1920
screen_height = 1080
pen_max_x     = 68840
pen_max_y     = 38720
pen_max_z     = 8191
resolution    = 5080
buttons       = 10
scrollbar     = 1
```

Could you add it to the config defaults please?

It take me a while to understand that those settings where necessary to make the tablet work...I was focused on monitor handling...Maybe you could emphasise the fact that the main thing to change in config is the tablet name :)

triantares, 2019-10-02:
Well, if you looked on the rear of your Kamvas pro 16 you might have seen the model number there.
It's [GT-133].
In time there will probably be tablets under the same name with different hardware and models so this is actually the best way.

joseluis, 2019-10-03:
Hi @sebRomeo thanks!

I'll add it, but I prefer to use the actual model name. Is this one yours? [GT-156](https://huion.uk/item/145-618/huion-kamvas-pro-16) it has 6 buttons,

@triantares What do you mean? the [tablet_gt133] already on config.ini has different values.

sebRomeo, 2019-10-04:
Ok thanks seems that GT-156 is exactly kamvas pro 16 :sweat_smile:
Maybe this could appear somewhere in the readme, or in some support section (which will also improve SEO when we search for linux driver kamvas pro 16 :smile_cat: ) ?

commit 9b917b19 (2019-10-04): add tablet gt-156 kamvas 16 pro; fix #37
closed 2019-10-04 by joseluis