joseluis/huion-linux-drivers #41: Script disables tablet

TheChertila, 2019-10-12:
## System Information
Tablet: HUION H420
Monitors: HDMI-0 FullHD & DVI-D-0 FullHD which are arranged
Kernel: 4.12.14-lp151.28.16-default

## Symptoms
* My tablet is H420, which is not supported yet.
* Script has disabled my tablet completely
* Same problem as #40 .

## I have:
- [x] installed all dependencies as it said in README
- [x] added extra code to Xorg
- [x] changed `config.ini` up to my system

There's output of `xinput` before running script, script output and `xinput` output after script fails.

```
chertila@openSUSE-PC ~/sources/huion-linux-drivers-master $ xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ COMPANY  USB Device                       id=15   [slave  pointer  (2)]
⎜   ↳ COMPANY  USB Device                       id=16   [slave  pointer  (2)]
⎜   ↳ COMPANY USB Device                        id=19   [slave  pointer  (2)]
⎜   ↳ COMPANY USB Device                        id=20   [slave  pointer  (2)]
⎜   ↳ HUION H420 Consumer Control               id=10   [slave  pointer  (2)]
⎜   ↳ HUION H420 Pen                            id=12   [slave  pointer  (2)]
⎜   ↳ HUION H420 Pad                            id=13   [slave  pointer  (2)]
⎜   ↳ HUION H420 Mouse                          id=22   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ COMPANY  USB Device                       id=14   [slave  keyboard (3)]
    ↳ UVC Camera (046d:0804)                    id=17   [slave  keyboard (3)]
    ↳ COMPANY USB Device                        id=18   [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=21   [slave  keyboard (3)]
    ↳ COMPANY  USB Device                       id=23   [slave  keyboard (3)]
    ↳ COMPANY USB Device                        id=24   [slave  keyboard (3)]
    ↳ HUION H420 System Control                 id=8    [slave  keyboard (3)]
    ↳ HUION H420 Keyboard                       id=9    [slave  keyboard (3)]
    ↳ HUION H420 Consumer Control               id=11   [slave  keyboard (3)]
chertila@openSUSE-PC ~/sources/huion-linux-drivers-master $ sudo ./huion-tablet-driver.py 
Finding USB device. . . Done!
grabbed interface %d 0
grabbed interface %d 1
grabbed interface %d 2
Reading configuration. . . Done!
Setting up driver. . . Done!
        Tablet model name         DEBUG TABLET
        Buttons                   disabled (0)
        Scrollbar                 disabled (0)
        Notifications             disabled
        Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.12.14-lp151.28.16-default (x86_64 x86_64)
#1 SMP Wed Sep 18 05:32:19 UTC 2019 (3e458e0)

DEVICE: /dev/input/event0 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x4

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 004437", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
Unable to connect to X server
ERROR running the following comand:
        xinput list --short | grep "Tablet Monitor Pen"
RETURN CODE: 1
chertila@openSUSE-PC ~/sources/huion-linux-drivers-master $ xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ COMPANY  USB Device                       id=15   [slave  pointer  (2)]
⎜   ↳ COMPANY  USB Device                       id=16   [slave  pointer  (2)]
⎜   ↳ COMPANY USB Device                        id=19   [slave  pointer  (2)]
⎜   ↳ COMPANY USB Device                        id=20   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ COMPANY  USB Device                       id=14   [slave  keyboard (3)]
    ↳ UVC Camera (046d:0804)                    id=17   [slave  keyboard (3)]
    ↳ COMPANY USB Device                        id=18   [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=21   [slave  keyboard (3)]
    ↳ COMPANY  USB Device                       id=23   [slave  keyboard (3)]
    ↳ COMPANY USB Device                        id=24   [slave  keyboard (3)]
```

joseluis, 2020-04-28:
Hi, please try the latest version, and maybe a more recent kernel would be of help.

yoanndaly, 2020-07-01:
I have just solved a very similar issue with a Huion Kamvas 13 on Ubuntu 18.04. For me, this was not a driver issue but an installation error. I share the solution in case this may be helpful to others. 

First, here is my return in debug mode before solving the problem.

[issue41.txt](https://github.com/joseluis/huion-linux-drivers/files/4860101/issue41.txt)

The issue for me was that evdev was not installed. I missed an error message at installation with dpkg. So I turned to Synaptic for installation (this is my preferred installation program in ubuntu). In ubuntu 18.04, I needed to uninstall first several programs including xorg, fix various broken dependencies (that prevented a direct installation simultaneously with the uninstallation ), and then finally install xserver-xorg-input-evdev (after uninstalling several other programs and installing dependencies that synaptic had missed at first). Then, even without restart, the script finally worked fine, with this result (not in debug mode anymore). 

[issue41afterSolution.txt](https://github.com/joseluis/huion-linux-drivers/files/4860170/issue41afterSolution.txt)

However, I have got a strange issue after reboot because of all the uninstallations needed, the pointer of my touch-pad didn't work anymore and I had to install xserver-xorg-input-synaptics as explained
[here](https://askubuntu.com/questions/1041820/touchpad-not-working-on-ubuntu-18-04) in order to solve the problem. At the end, the driver is working just fine.

Thanks again to Joselius for this very nice driver.
Yoann
 
PS: To be thorough, here is the diagnostic that I carried out pointing out to an evdev issue.
After unpluging and pluging the tablet (so that hid-uclogic recognized the tablet again), I tried to follow the first steps of the script. I give results of the relevant part of `xinput list` before and after each step.

[xinputresults.txt](https://github.com/joseluis/huion-linux-drivers/files/4860313/xinputresults.txt)

So deactivation or reactivation of hid-uclogic destroys or recovers the pen detection, but after activation of uinput,  hid-uclogic was not able to recover the Pen anymore, and the resulting output for xinput was the same as  the one at the end of the python script. In the script, there was apparently only applications of evdev (via its python package) in between the activation of uinput and the use of xinput. I supposed that evdev was supposed to recover the Pen device in xinput, so I suspected that my problem came from evdev...