joseluis/huion-linux-drivers #43: [Help] Huion GT-133 / Kamvas pro 13 with Debian Buster

nioc, 2019-11-18:
Hello, i'm a noob in this domain, so maybe i forget do do something...

Installed prequisites: ok
```
sudo apt install xinput xserver-xorg-input-evdev python3-evdev python3-usb \
> xdotool libnotify-bin arandr python3-numexpr
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
arandr est déjà la version la plus récente (0.1.9-2).
libnotify-bin est déjà la version la plus récente (0.7.7-4).
python3-numexpr est déjà la version la plus récente (2.6.9-1).
python3-evdev est déjà la version la plus récente (1.1.2+dfsg-1+b10).
python3-usb est déjà la version la plus récente (1.0.2-1).
xdotool est déjà la version la plus récente (1:3.20160805.1-4).
xinput est déjà la version la plus récente (1.6.2-1+b1).
xserver-xorg-input-evdev est déjà la version la plus récente (1:2.10.6-1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
```
And `/etc/X11/xorg.conf.d/evdev-tablet.conf `set too.

Configuration: laptop running Debian 10 (Xorg not wayland), screen eDP-1 1920x1080,
Device: Huion GT-133, I tried to set it in config, but nothing happened, so I tried debug tablet : 
```
Finding USB device. . . Done!
grabbed interface %d 0
grabbed interface %d 1
Reading configuration. . . Done!
Setting up driver. . . Done!
	Tablet model name         DEBUG TABLET
	Buttons                   disabled (0)
	Scrollbar                 disabled (0)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     ENABLED
	Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.19.0-6-amd64 (x86_64 )
#1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11)

DEVICE: /dev/input/event6 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 170053", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
⎜   ↳ Tablet Monitor Pen 170053               	id=17	[slave  pointer  (2)]
Calibrating. . . Done!

Huion Kamvas driver should now be running. . .

(Input from the tablet will ONLY be used to print it out)
```
Nothing... I have to exit with CtrlC:
```
^CTraceback (most recent call last):
  File "./huion-tablet-driver.py", line 755, in <module>
    main.run()
  File "./huion-tablet-driver.py", line 35, in run
    main_loop()
  File "./huion-tablet-driver.py", line 313, in main_loop
    data = main.dev.read(main.endpoint.bEndpointAddress, main.endpoint.wMaxPacketSize)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 851, in intr_read
    timeout)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 933, in __read
    timeout)
KeyboardInterrupt

```
Any idea what went wrong?

When I restart tablet (ie without starting script), it works well in duplicated (1+1 monitor) mode but with an offset in 2 monitors :disappointed:

nioc, 2019-11-18:
Seems to be the same behavior #34

lululock71, 2020-01-09:
I've encountered a similar behavior with my Kamvas Pro 13 but it doesn't seem to be linked to buttons. On mine, if I just launch the script after plugging it, I get no reaction at all. I need to disconnect the USB cable to get it to work again.

My solution :
Before launching the script, just hoover the stylus over the tablet to make the cursor move. Then, when you launch the script, it just works. I dunno why but it looks like some driver calibration error...

I've submitted this issue but nobody responded so far...
[https://github.com/joseluis/huion-linux-drivers/issues/45](url)

For the dual monitor offset, I haven't found a solution yet, I have a similar issue...