joseluis/huion-linux-drivers #45: Stylus is disabled if it hasen't been used before starting the script...

lululock71, 2019-11-21:
Hi,
It might sounds weird but I noticed that with my Kamvas Pro 13:

When I plug the tablet, if I don't use the stylus at least once before launching the driver (like hover over the screen and make the cursor move), the stylus don't work anymore after the driver is started.

If i exit the script, it doesn't work anymore. I need to reboot the system (or maybe restart xorg should do it) in order to get the stylus functions back.

I don't know if it does that with other tablets, since I don't have another Huion model to test it out.

Queatz, 2020-04-04:
Experiencing something similar, but maybe slightly different.  It seems to work after plugging it in, until I run the script.  But I need to remap the pointer by running the below command, and then everything is working.
`xinput set-prop 12 --type=float "Coordinate Transformation Matrix" 0.25 0 0.0 0 1.0 0.0 0 0 1`

joseluis, 2020-04-28:
I can't replicate this with my tablet... It seems to happen only for some tablets.