joseluis/huion-linux-drivers #47: Huion Kanvas 16 Pro - Calibration values

t0cc, 2019-12-05:
I did play around with calibrating my tablet yesterday. I came up with slightly different values. Maybe they help you, too. I'm not sure if this is just my setup.
```
[tablet_gt156v2]
model_[tablet_gt156v2]
model_name    = GT-156 v2
screen        = true
screen_width  = 1920 
screen_height = 1080 
# pen_max_x     = 68834
pen_max_x     = 68665
# pen_max_y     = 38720
pen_max_y     = 38650
pen_max_z     = 8191
# resolution    = 5080
resolution    = 5050
buttons       = 14  
scrollbar     = 1   
```

lululock71, 2019-12-05:
I personally have some negative values. Do you think it affects something?

Because I feel like the calibration doesn't work, I have a 3mm offset when the stylus is in the top left and top right corners but it looks okay in the middle of the screen...

This annoys me a bit but I got used to.

I own the 13 inch variant.

t0cc, 2019-12-08:
Play around with those values I changed. Then after some time you will see how they behave. Without real calibration by the driver it can take quite some time to find a good working set of values. 

It has to do with finding the right width/height value in conjunction with the resolution.  Unfortunatley, there are no x/y-Offsets so far. Not sure if they are missing or just not needed.

joseluis, 2020-04-28:
Hi @t0cc and @lululock71  have you tried enabling and modifying the calibration data values at the beginning of the config.ini file?:
```
# Calibration Data
enable_calibration      = false
calibrate_min_x         = 250
calibrate_max_x         = 95250
calibrate_min_y         = 300
calibrate_max_y         = 53400
```

lululock71, 2020-04-29:
Yes, those are the values I experimented with. After hours of struggle, I haven't found settings that would completely remove this offset.

Which is the most weird, is that when I test for the offset, it is not consistent over the whole screen.
I do a "9-points" test all over the screen and I noticed that the most accurate position is in the down left corner and the middle point are almost aligned with the pen tip. But other points can have a offset up to 3mm on the Y axis (cursor to the left of the pen tip) and approx 1mm on the X axis (cursor bellow the pen tip)... Maybe some picture can help, I can provide some if you need.

The offset seems not to be linear, or my tablet has a malfunction ? I managed to calibrate it correctly when testing it under Windows once. Maybe there's some stuff to reverse engineer from the Huion driver for Windows ? I might ask Huion if they could give us some more information or at least give us some clues...

I used the tablet with default settings since and get used to the offset... This is kinda annoying because you shouldn't have to do that.