joseluis/huion-linux-drivers #49: Map input to graphics monitor and calibrate using `xinput map-to-output`

shervinsahba, 2019-12-10:
I had difficulty calibrating my tablet via `xinput_calibrator`, since the calibration screen always appeared on my primary monitory instead of the graphic's tablet monitor. I wanted to confine the pen to my tablet's monitor. One solution appears to be `xinput map-to-output <device> <output>`. The result confines the pen and appears calibrated on my Kamvas Pro 12.

I added this functionality to the main script by adding the below code to `huion-tablet-driver.py`. This script may need to be touched up for generality, especially if you're using anything other than the [monitor_1] setting. It assumes your primary display comes up first in `xrandr` and that the graphics monitor comes up second. 

This works with the automatic loader #35 by @yayfortrees, but you need to change the service `Type=idle`, else I believe the system tries to prematurely run xrandr before the second display is live.

```
XINPUT_DEVICE_KEYWORD = "Tablet Monitor Pen"

def calibrate_mapping(xinput_device_keyword=XINPUT_DEVICE_KEYWORD):

    try:
        # Locate tablet pen device number
        xinput_device = os.popen("xinput | grep '%s' | cut -f 2" % xinput_device_keyword).read()
        xinput_device = xinput_device.split('\n')[0][3:]   # chooses the first instance of keyword, strips off "id=""

        # Locate output device names
        xoutput_devices = os.popen("xrandr --listactivemonitors | sort | cut -f 6 -d ' '").read()
        xoutput_devices = list(filter(None,xoutput_devices.split('\n')))    
		# assumes the first device is the main monitor and the second is the graphics monitor

        # Calibrate by mapping input to graphics pad output
        os.system("xinput map-to-output %s %s" % (xinput_device, xoutput_devices[1]))
        print()
        sys.stdout.write("Calibrating graphics monitor via xinput mapping. . . ")
    except:
        print("Error with calibrate_mapping")

    print("Done!") 
    print("xinput_device = %s" % xinput_device)
    print("xoutput_device = %s" % xoutput_devices[1]) 
    print("xoutput_mainscreen = %s" % xoutput_devices[0]) 
```

where the `calibrate_mapping` function is added to the `main()` class before the finishing loop.

```
class main():
    """
    """
    settings = {'pen_device_name':'Tablet Monitor Pen' # must be defined here
                + strftime(" %H%M%S", gmtime())}       # for pressure to work
    dev = None
    endpoint = None
    vpen = None
    current_menu = None

    def run():	
        find_usb_device()
        read_config()
        #prepare_driver()
        setup_driver()
        calibrate()
        multi_monitor()
        calibrate_mapping()        
        main_loop()
```

Feedback appreciated!

Fatih20, 2020-01-18:
I know you're talking about pen display, but can this be used to confine pen tablets to only one monitor in a multi monitor setup? @shervinsahba

shervinsahba, 2020-01-19:
@Fatih20 Let's see if I understand. Do you want to calibrate and confine the pen to some _non-tablet_ monitor in your setup? In that case, not necessarily. 

**As it stands, the function maps the pen to the second active monitor in `xrandr`** (my setup had my primary monitor plus my tablet's built-in monitor as the secondary, so I was hacking to solve that setup). 

How can you choose a different monitor, then? 

You want to identify the argument of your desired monitor in the `xoutput_devices` list. Your primary monitor will probably be `xoutput_devices[0]`, secondary `xoutput_devices[1]`, and so on. If you're unsure which monitor is which, running `xrandr --listactivemonitors | sort` in a terminal may give some insight.

Let's say you found out that your desired monitor is `xoutput_devices[j]`. Then change `xoutput_devices[1]` to `xoutput_devices[j]` in this line.

`os.system("xinput map-to-output %s %s" % (xinput_device, xoutput_devices[j]))`

I think that should work, but I need to test it.

shervinsahba, 2020-01-19:
You also may need to set `XINPUT_DEVICE_KEYWORD = "Tablet Monitor Pen"` to match whatever your pen appears as in the output of `xinput`.

Fatih20, 2020-01-19:
@shervinsahba No no, what I mean is to use a pen tablet (H950P, H640P, etc.) and confine the drawing area of those device to only one monitor among many in a multi-monitor setup.

shervinsahba, 2020-02-21:
@Fatih20 Sorry, I don't think I understand, because your issue _still_ seems to be answered by what I described in my response. Maybe someone else can chime in if I'm missing the point.

Have you tried those steps I mentioned to find your desired monitor with `xrandr --listactivemonitors` to see if it works? If you have more than two monitors, you'll need to set it manually in the code.