joseluis/huion-linux-drivers #50: Pen positioning changing when tilting the pen

FrozenInc, 2019-12-17:
The cursors position on the screen changes when the pen is tilted. Luckily I was able to find a solution. The values that I have used for '''pen_sensor_length''' works for my specific pen and tablet: Huion Kamvas Pro 13, linux sees it as GT-133.

```
# compensate positions with tilt
pen_sensor_length = 100
if(TILT_X > 100):
    pen_sensor_length = 300
    X = X + int(pen_sensor_length*atan(TILT_X*1.40625))
else: 
    X = X - int(pen_sensor_length*atan(TILT_X*1.40625))
if(abs(TILT_Y) > 100):
    pen_sensor_length = 50
    Y = Y - int(pen_sensor_length*atan(abs(TILT_Y*1.40625)))
else: 
    pen_sensor_length = 50
    Y = Y + int(pen_sensor_length*atan(abs(TILT_Y*1.40625)))

```

This is not the best solution, as it needs to be implemented to work for all tablets, but the math is there and it works almost perfectly. I used 1.40625 as a multiplier as the tilt goes from 0 to 255, but degrees are from 1 to 360, so i had to bind 1 tilt to 1.40625 degrees. The pen_sensor_length are arbitrary, and every single tablet will be needed to be calibrated by hand.

lululock71, 2020-01-09:
It works for me, but I had to edit a bit this code and re-calibrate my cursor...
I still have a offset but it is almost imperceptible, especially during normal use.

For unknown reasons, I had to add `math.` before `atan(...)` because it generated an error.
My tablet is also a Kamvas Pro 13.

The code must be pasted at line 452 :
`# compensate positions with tilt
                pen_sensor_length = 100
                if(TILT_X > 100):
                    pen_sensor_length = 300
                    X = X + int(pen_sensor_length*math.atan(TILT_X*1.40625))
                else: 
                    X = X - int(pen_sensor_length*math.atan(TILT_X*1.40625))
                if(abs(TILT_Y) > 100):
                    pen_sensor_length = 50
                    Y = Y - int(pen_sensor_length*math.atan(abs(TILT_Y*1.40625)))
                else: 
                    pen_sensor_length = 50
                    Y = Y + int(pen_sensor_length*math.atan(abs(TILT_Y*1.40625)))

`

Fatih20, 2020-01-13:
@FrozenInc What do you mean calibrated by hand?
Perhaps it's a good idea to make a pull request for this.

joseluis, 2020-04-28:
PR #66 added tilt support, please try if it works for you

lululock71, 2020-04-29:
Hi,
Tested on the newest version. The cursor still changes position when tilting the pen.
By the way, this also occurs when the script is not running, suggesting that this is somehow generated by the tablet itself.

The fix showed above reduced the offset but it is still present.

I think that this issue can be related to the calibration issue mentioned in issue https://github.com/joseluis/huion-linux-drivers/issues/47

popcko, 2022-03-10:
Hi! this works on kamvas pro 16 but can you make it so it will follow the pen while drawing the line? This will save me, Thank you!