joseluis/huion-linux-drivers #51: Huion HS64 doesnt work

5H4D0WILA, 2019-12-24:
Finding USB device. . . Done!
Reading configuration. . . Done!
Setting up driver. . . Done!
        Tablet model name         DEBUG TABLET
        Buttons                   disabled (0)
        Scrollbar                 disabled (0)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 5.0.0-37-generic (x86_64 x86_64)
#40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019

DEVICE: /dev/input/event21 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 153024", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
ERROR running the following comand:
        xinput list --short | grep "Tablet Monitor Pen"
RETURN CODE: 1

that is my [tablet_debug] output

joseluis, 2020-04-28:
I can't see what's wrong. Can you try the latest version?

Combinacijus, 2020-06-21:
Huion HS64 works for me well