joseluis/huion-linux-drivers #55: HUION GT190

blasco, 2020-01-12:
I got the Huion GT190 almost working. I would like some guidance on how obtain the different parameters:

```
[tablet_gt190]
model_name    = GT-190
screen        = true
screen_width  = 1400
screen_height = 900
pen_max_x     = 0
pen_max_y     = 0
pen_max_z     = 0
resolution    = 0
```

What are `pen_max_x` and in what units and how can I figure out that value? Same thing with y,z and resolution. 

Apart from that the tablet seems to be working, but only for the X axis (any idea what could be making the pen to react to the X axis but not the Y axis, could it just be the wrong pen_max_y numbers?

Thank you for the guidance and support!

joseluis, 2020-04-28:
Hi, sorry for the long delay in answering.

Now you can see those numbers when you start the driver either with the `debug_mode = true` option, or the `current_tablet = [tablet_debug]`, in `config.ini`.
E.g. for my tablet the right numbers are in the `Params block #2`:

```
$ sudo ./huion-tablet-driver.py
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 37540
    |                          Max Y: 21120
    |                   Max pressure: 8191
    |                     Resolution: 2000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 95352
    |                          Max Y: 53645
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M167_170623
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name         DEBUG TABLET
	Buttons                   disabled (0)
	Scrollbar                 disabled (0)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    disabled
```

Please try it with the latest version

MimosaBoo, 2022-10-10:
Hello, 
I followed the steps to make the Huion GT190 work, everything worked except the orientation of the cursor which is reversed, when I go down it goes up, when I go left it goes right ... 
I couldn't find a solution. 
I've been using my huion q11 v2 tablet with a script for linux, but the cursor reversal problem is also present. 

Help!    

(Normally the tablet is detected in the internal "graphic tablet" software of linux, it is here that I indicated the very first time the left-handed orientation (button on the left) but the tablet is not detected)