joseluis/huion-linux-drivers #56: Support for Huion-h1161

mounika-v, 2020-01-12:
Hello, I am using a Huion-H1611. It is not in the list and this is the stack-trace I've received. Thanks in advance.

`Finding USB device. . . Done!
grabbed interface %d 0
grabbed interface %d 1
Reading configuration. . . Done!
Setting up driver. . . Done!
	Tablet model name         DEBUG TABLET
	Buttons                   disabled (0)
	Scrollbar                 disabled (0)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.15.0-72-generic (x86_64 x86_64)
#81-Ubuntu SMP Tue Nov 26 12:20:02 UTC 2019

DEVICE: /dev/input/event6 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 183507", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
ERROR running the following comand:
	xinput list --short | grep "Tablet Monitor Pen"
RETURN CODE: 1
`

Kenward594, 2020-03-28:
Hi!  I was hoping for the same.  Any recent work in this direction?  I need a tablet for my teaching work (with the virus going crazy...) and thought this would work well, but I see no list of specific tablets supported.  ??

joseluis, 2020-04-28:
Please take a look to [this thread on the digimend kernel drivers repo](https://github.com/DIGImend/digimend-kernel-drivers/issues/314) regarding support for that tablet.