joseluis/huion-linux-drivers #67: GT-191 broken after reinstall

BugHunterPhilosopher, 2020-03-21:
Good afternoon,

First of all, thanks a lot for your hard work on the driver and on the issues! I have used it for almost one year flawlessly on my Huion GT191, it's a breeze.

Unfortunately, I changed my PC and had hence to reinstall. I now have this set-up:

- PopOS 19.10 (Ubuntu-based)
- kernel 5.5.3 (compiled by myself)
- uclogic, master, freshly compiled
- Huion driver, master.

I have no error in the terminal, but no input either. Can you please help?

Thanks a lot & Regards

BugHunterPhilosopher, 2020-03-21:
I just want to let you know that I didn't forgot the xorg files. :-)

joseluis, 2020-04-28:
Hi, I don't know what may be wrong, but it seems to be related to #68  and Ubuntu 19.10

closed 2020-04-28 by joseluis