joseluis/huion-linux-drivers #68: No pointer at all

agarrubio, 2020-03-24:
Script reaches (Input from the tablet will ONLY be used to print it out), but coordiantes of of pointer are not shown, and I have to interrupt with Ctrl+C.
```
orquita% sudo ./huion-tablet-driver.py
Finding USB device. . . Done!
Reading configuration. . . Done!
Setting up driver. . . Done!
	Tablet model name         DEBUG TABLET
	Buttons                   disabled (0)
	Scrollbar                 disabled (0)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.15.0-20-generic (x86_64 x86_64)
#21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018

DEVICE: /dev/input/event3 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 034322", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
⎜   ↳ Tablet Monitor Pen 034322               	id=8	[slave  pointer  (2)]

Huion Kamvas driver should now be running. . .

(Input from the tablet will ONLY be used to print it out)
```

Queatz, 2020-04-04:
This started happening to me as well.  I believe upgrading to Ubuntu 19.10 stopped the pen input from working.  Currently trying different things to get input back...

BugHunterPhilosopher, 2020-05-01:
Joseluis,

How can we help, please?

joseluis, 2020-05-01:
@BugHunterPhilosopher for one, we should investigate whether something has changed in the latest ubuntu version that affects the evdev input values...

BugHunterPhilosopher, 2020-05-01:
OK, I'll try & do that. I use the latest Pop! OS, by the way (Ubuntu-based).

oceanedbs, 2020-07-21:
Did someone find a solution ? I have the same problem ... :)