joseluis/huion-linux-drivers #69: HUION HS64 pressure detection sensitivity on Linux

kaphula, 2020-04-02:
Hello,

I have HUION HS64 tablet and I tried out this driver. It works alright but the pressure detection is not very good on Linux compared to Windows. The thing that bothers me is that I have to apply too much force before the drawing activates at all. The pen's own weight is not enough to activate the drawing. I can't just swipe the pen on gently on top of my tablet to create very light lines. I use Krita as my painting program and playing around with the pressure settings does not help.

Is it possible to adjust this kind of sensitivity setting with this driver?

lululock71, 2020-04-29:
You can set a custom pressure curve into Krita to get the desired sensitivity.
The setting is under "Krita Configuration/tablet settings"
(it might not be exactly called like that since I use Krita in French)

I personally kept the linear default setting and it feels right to me.

It's usually a better idea to adjust a pressure curve directly into each program. It can be annoying if you use lots of programs but you can adjust it to each use case and not just have a global setting. Sure, it would be cool if we could set a custom curve directly into the driver but it's not a priority since you can do exactly what you want with the options embedded in each program in which a graphic tablet can be used.

Hope it solved your issue.

kaphula, 2020-05-01:
I got the pressure settings to acceptable levels eventually with the Krita's pressure curve tool. Thanks!

closed 2020-05-01 by kaphula