joseluis/huion-linux-drivers #70: xinput error, unable to find device Tablet Monitor Pen 083724

ameanie, 2020-04-03:
Hi, sorry if this isn't the correct information to provide, very new to Linux so I'm still figuring my way around things.

`Finding USB device. . . Done!
grabbed interface %d 0
grabbed interface %d 1
Reading configuration. . . Done!
Setting up driver. . . Done!
	Tablet model name         GT-156
	Buttons                   ENABLED (8)
	Scrollbar                 ENABLED (1)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    Enabled (1920x1080)
	Current Monitor Setup     [monitor_2]
	Multi Monitor Setup       ENABLED
	Calling xrandr            disabled

Setting up multiple monitors. . . 
Running xinput. . . 
unable to find device Tablet Monitor Pen 083724
ERROR running the following comand:
	xinput set-prop "Tablet Monitor Pen 083724" --type=float "Coordinate Transformation Matrix" 0.42857142857142855 0 0.5714285714285714 0 0.75 0.0 0 0 1
RETURN CODE: 1
`

I've both commented out the call for the Tablet Monitor Pen area in the .py file, and edited my config.ini to not have debug enabled but it's still giving me this error.

The tablet pen, nor the buttons on my tablet (Kamvas Pro 16) are responsive.

Queatz, 2020-04-04:
You might want to check this part of the readme: https://github.com/joseluis/huion-linux-drivers#xorg-extra-code-

ameanie, 2020-04-04:
> You might want to check this part of the readme: https://github.com/joseluis/huion-linux-drivers#xorg-extra-code-

Hi, sorry I kept trying to do this but didn't have root access and was kind of concerned on how to get access. I just did and added the file and now when i run the .py file I get this :

`Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 3, in <module>
    import usb.core, usb.util
ModuleNotFoundError: No module named 'usb'
`

joseluis, 2020-04-28:
Hi, You must install the pyusb module. It's different depending on your system. You can search in google for the error and follow the instructions to install it.

For example in Ubuntu based distributions you'd have to install the `python3-usb` package.

ameanie, 2020-04-28:
> Hi, You must install the pyusb module. It's different depending on your system. You can search in google for the error and follow the instructions to install it.
> 
> For example in Ubuntu based distributions you'd have to install the `python3-usb` package.

Hi thank you for the reply!

I've installed that according to your original guide I believe,

I ran "$ sudo apt install python3-usb"

```
Building dependency tree       
Reading state information... Done
python3-usb is already the newest version (1.0.2-1).
0 upgraded, 0 newly installed, 0 to remove and 94 not upgraded.
```
I actually haven't been on Linux since the last time I was trying to troubleshoot this, but since then it seems like my tablet is responsive now, the mouse moves and I can click which is great! But it doesn't seem to line up with the tablet pen at all i just have to scroll the arrow like you would with a mouse, and I'm not sure how to align it. It also doesn't let me move the pen/mouse pointer all the way to the end of the tablet.

I tried running the python driver file (forgive me if this was a mistake, i'm still new to this) and received this error :
```

Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 3, in <module>
    import usb.core, usb.util
ModuleNotFoundError: No module named 'usb'

```

ameanie, 2020-04-28:
`#!/usr/bin/env python3

import usb.core, usb.util
import sys
import os.path
import platform
from evdev import UInput, ecodes, AbsInfo, InputDevice, list_devices
import subprocess as sp
import math
import numexpr
from configparser import ConfigParser, ExtendedInterpolation
from time import gmtime, strftime, sleep

MENU = {}

# -----------------------------------------------------------------------------
class main():
    """
    """
    settings = {'pen_device_name':'Tablet Monitor Pen' # must be defined here
                + strftime(" %H%M%S", gmtime())}       # for pressure to work
    dev = None
    endpoint = None
    vpen = None
    current_menu = None

    def run():
        find_usb_device()
        read_config()
        #prepare_driver()
        setup_driver()
        calibrate()
        multi_monitor()
        main_loop()

# -----------------------------------------------------------------------------
def find_usb_device():
    """
    """
    sys.stdout.write("Finding USB device. . . ")

    device_ids = [(0x256c, 0x006d), (0x256c, 0x006e)]
    for cur_dev_id in device_ids:
        main.dev = usb.core.find(idVendor=cur_dev_id[0], idProduct=cur_dev_id[1])
        if main.dev: break

    if not main.dev:
        print("Error, Could not find device, maybe already opened?",
            file=sys.stderr)
        sys.exit(1)
    else:
        print("Done!")

    for cfg in main.dev:
        for i in cfg:
            for e in i:
                if not main.endpoint:
                    main.endpoint = e
            if main.dev.is_kernel_driver_active(i.index):
                main.dev.detach_kernel_driver(i.index)
                usb.util.claim_interface(main.dev, i.index)
                print("grabbed interface %d", i.index)

    main.endpoint = main.dev[0][(0,0)][0]

# -----------------------------------------------------------------------------
def prepare_driver():
    """
    This is necessary for now.
    See https://github.com/benthor/HuionKamvasGT191LinuxDriver/issues/1
[... 643 lines cut ...]
                    MENU[section][n] = config.get(
                        section, btn).strip()
                else:
                    MENU[section][n] = ""

            # scrollbar
            if main.settings['scrollbar']:
                try:
                    MENU[section]['scroll_up'] = config.get(section, 'su').strip()
                except:
                    MENU[section]['scroll_up'] = ""
                try:
                    MENU[section]['scroll_down'] = config.get(section, 'sd').strip()
                except:
                    MENU[section]['scroll_down'] = ""

    main.current_menu = main.settings['start_menu']

    print("Done!")

# -----------------------------------------------------------------------------
if __name__ == '__main__':
    main.run()`

I'm pasting my huion-tablet-driver.py just in case it's needed.

joseluis, 2020-04-29:
Hi, well the py file wasn't really needed, but I can see you're using an old version.

Please try the latest version of both the huion-tablet-driver.py and config.ini (changed for your tablet)

ameanie, 2020-04-29:
I've updated the two files, as well as changing the config for my tablet and receive more or less the same error : 

```

Traceback (most recent call last):
  File "./huion-tablet-driver.py", line 8, in <module>
    import usb.core
ModuleNotFoundError: No module named 'usb'
```

Just with the line number changing, not really sure how to troubleshoot this on my own sadly sorry.
I've made sure to attempt the install again and there doesn't seem to be any changes or updates.

I also searched the specific error and it seems like you said that pyusb module isn't installed, so I went and installed pyusb separately again using : https://pypi.org/project/pyusb , with 
`sudo apt install pyusb`
and then
`pip install pyusb` 
(both seemed to install?)
I've also installed libusb as that seems to be a prerequisite?

I attempted to run the driver between each step, I've also restarted my PC as well just in case. Still having the same issues though, sadly.

joseluis, 2020-04-29:
Oh well, pyusb is not what is needed, that's an abandoned project for python2, you can safely uninstall it.

Try with: `sudo apt install python-usb python3-usb `

ameanie, 2020-04-29:
I installed and it seemed to complete but I still receiving the same error, restarted after as well.

`python-usb is already the newest version (1.0.2-1).
python3-usb is already the newest version (1.0.2-1).
The following packages were automatically installed and are no longer required:
  linux-headers-5.3.0-7625 linux-headers-5.3.0-7625-generic
  linux-image-5.3.0-7625-generic linux-modules-5.3.0-7625-generic
  linux-modules-extra-5.3.0-7625-generic python-certifi python-chardet
  python-idna python-openssl python-requests python-urllib3
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

joseluis, 2020-04-30:
Thanks, don't worry :smile: 

That is weird. To make import test call the python3 shell from the terminal, and try to import the usb module like this (the commands you must write are shown after the `>>>` prompt):
```
$ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
>>> 
```

if you dont have the module will say something like this:
```
>>> import thiswillfail
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'thiswillfail'
>>> 
```

If can load the `usb` module, try loading the `usb.core` one. If that one fails then it must be trying to load an old version have an old version of it, and you have your system missconfigured somehow...

ameanie, 2020-05-02:
Hi, it fails when I call usb.core :
```

Python 3.8.2 (default, Apr 29 2020, 18:07:50) 
[GCC 9.2.1 20191008] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'usb'

```

joseluis, 2020-05-03:
And this one is another reason why I want to recreate this project in Rust. I'm fed up python, its import system and its unconsistency between installations...

joseluis, 2020-05-03:
BTW I also recommend to use python3.6. At least for me this script doesn't work well with newer versions.

ameanie, 2020-05-03:
Well, I followed these instructions (https://forum.manjaro.org/t/how-to-downgrade-python-version-to-3-6-7/83039/5) because I couldn't install Python3.6 with any luck since it seemed to not accept it with new versions installed.

Now I can't open Terminal at all, lol. I think I'm going to give up on this for now, maybe I'll reinstall the OS and try again at a later time.