joseluis/huion-linux-drivers #72: Huion 156 v2 debug data

tr37ion, 2020-04-27:
Here is the debug output I get from this missing device. I inserted the lowest and max X7Y coordinates I could determine.

```
Finding USB device. . . Done!
Reading configuration. . . Done!
Setting up driver. . . Done!
        Tablet model name         DEBUG TABLET
        Buttons                   disabled (0)
        Scrollbar                 disabled (0)
        Notifications:            ENABLED
                for buttons       ENABLED
                for scrollbar     disabled
        Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        < DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
SYSTEM: Linux 5.7.0-1-MANJARO (x86_64 )
#1 SMP PREEMPT Tue Apr 21 20:48:43 UTC 2020

DEVICE: /dev/input/event31 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 133803", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
⎜   ↳ Tablet Monitor Pen 133803                 id=22   [slave  pointer  (2)]

Huion Kamvas driver should now be running. . .

(Input from the tablet will ONLY be used to print it out)

...
08 00 2b 01 00 00 00 00 00 00 00 00 | X:00000 Y:00000 PRES:0000 TILT_X:000 TILT_Y:000
...
08 00 e3 f1 40 97 00 00 00 00 22 2e |  X:68840 Y:38720 PRES:0000 TILT_X:034 TILT_Y:-46
```

tr37ion, 2020-04-27:
I added the following data to the scrypt and it works ok. Except tilting is not recognized while I actually get tilting data output from the tablet. I added some example data below `config.ini` settings

```
[menu_simple_14v2]
button 0 = key Tab           # hide interface
button 1 = key r             # rect select (gimp) & pick layer (krita)
button 2 = key ctrl+x        # cut
button 3 = key ctrl+c        # copy
button 4 = key ctrl+v        # paste
button 5 = key ctrl+z        # undo
button 6 = key ctrl+y        # redo (gimp)
button 7 = key ctrl+shift+z  # redo (krita)
button 8 = key 4            # turn left (krita)
button 9 = key 6            # turn right (krita)
button 10 = 
button 11 = 
button 12 = 
button 13 = 
...
[tablet_gt156v2]
model_name    = GT-156 v2
screen        = true
screen_width  = 1920
screen_height = 1080
pen_max_x     = 68840
pen_max_y     = 38720
pen_max_z     = 8191
resolution    = 5080
buttons       = 14
scrollbar     = 1
...
[monitor_3+1M]
# 3 monitors arranged horizontally and the tablet below the middle one
# screens widths and heights
screen_1W = 2560
screen_1H = 1440
screen_2W = 2560
screen_2H = 1440
screen_3W = 2560
screen_3H = 1440
screen_4W = 1920
screen_4H = 1080

# global geometry
total_screen_width   = ${screen_1W} + ${screen_2W} + ${screen_3W}
total_screen_height  = ${screen_1H} + ${screen_4H}
tablet_offset_x      = ${screen_1W}
tablet_offset_y      = ${screen_1H}

xrandr_output1 = --output DP-0 --mode ${screen_1W}x${screen_1H} --pos 0x0 --rotate normal
xrandr_output2 = --output DP-2 --mode ${screen_2W}x${screen_2H} --pos ${screen_1W}x0 --rotate normal
xrandr_output3 = --output DP-4 --mode ${screen_3W}x${screen_3H} --pos 5120x0 --rotate normal
xrandr_output4 = --output HDMI-0 --mode ${screen_4W}x${screen_4H} --pos ${tablet_offset_x}x${tablet_offset_y} --rotate normal
xrandr_args    = ${xrandr_output1} ${xrandr_output2} ${xrandr_output4} ${xrandr_output3}
```

Some tilting debug output max data I found and the orientation of the pen laying down to one of the four compass points.

```
08 00 0c 7c 71 49 00 00 00 00 38 00 | X:31756 Y:18801 PRES:0000 TILT_X:056 TILT_Y:000 --> East
08 81 21 8a 37 5c 02 00 00 00 c4 00 | X:35361 Y:23607 PRES:0002 TILT_X:196 TILT_Y:000 --> West
08 00 e7 80 c6 56 00 00 00 00 f9 3b | X:32999 Y:22214 PRES:0000 TILT_X:249 TILT_Y:-59 --> North
08 00 e8 73 4b 28 00 00 00 00 01 ca | X:29672 Y:10315 PRES:0000 TILT_X:001 TILT_Y:-202 --> South
```

The range seems to be roughly:
```
WEST: 196 to 255 <-- TILT_X: 0 --> EAST: 0 to 060
NORTH: -54 to 0 <-- TILT_Y: 0 --> SOUTH: -255 to -202
```

What is strange that tilting the pen from **west** to **east** it goes from 196 to 255 then suddenly jumps to 0 and then counts on form 0 to 56 reaching **east**. Similar effect with **north** to **south** tilting as it looks tilting the pen to south counts the opposite way.

joseluis, 2020-04-28:
Hi, thank you for the information. I've just merged a PR adding tilt support (#66).

Please try the latest version, you'll have to add the folloing extra values to your `[tablet_gt156v2]` section:
```
pen_min_tilt  = -60
pen_max_tilt  = 60
```

tr37ion, 2020-04-29:
With Krita it seems that the cursor's position adds/subs an offset relative to the pen depending on the degree of tilting it. I'm using the Krita brush `Pencil-5 Tilted`. Testing it with Krita's **Tablet Tester** in settings dialog doesn't list tilting values at all. Maybe I miss something.

tr37ion, 2020-08-16:
Nice, tilt seems to work fine with this tablet now. I did a first test with Krita and brush `Pencil 5 - Tilted`. Directions are fine now, too. As tablet values I set:

```
[tablet_gt156v2]
model_name    = GT-156 v2
screen        = true
screen_width  = 1920
screen_height = 1080
pen_max_x     = 68665
pen_max_y     = 38650
pen_max_z     = 8191
resolution    = 5080
# resolution    = 5050
buttons       = 14
scrollbar     = 1
pen_min_tilt  = -60
pen_max_tilt  = 60
```