joseluis/huion-linux-drivers #73: uclogic-probe/uclogic-decode : No such file or directory

lululock71, 2020-04-29:
Hi,
I've downloaded the latest version but when I launch it, I've got this issue :
```

sudo python '/home/[...]/huion-tablet-driver.py'

Finding USB device. . . Done!
Reading configuration. . . Done!
hid_uclogic            32768  0
/bin/sh: /usr/local/bin/uclogic-probe: Aucun fichier ou dossier de ce type
/bin/sh: /usr/local/bin/uclogic-decode: Aucun fichier ou dossier de ce type
Preparing driver. . . ERROR running the following comand:
	"/usr/local/bin/uclogic-probe" "3" "2" | "/usr/local/bin/uclogic-decode"
RETURN CODE: 127

```

The older version still works.

Maybe I'm missing something about the dependencies but it worked before... _uclogic-tools_ is the only uclogic related package available for Arch.

joseluis, 2020-04-29:
For now you can write in terminal: `which uclogic-probe` in order to find out the path of uclogic, and you'll have to update the `uclogic_bins` settings in `config.ini` file. E.g.:

```
$ which uclogic-probe
/usr/bin/uclogic-probe
```
`config.ini`:
```
# Miscellaneus
uclogic_bins            = /usr/bin
```

I should probably make that part more automatic...

lululock71, 2020-04-29:
Thank you, it works now !

What I don't get, is why the old version worked...

closed 2020-04-29 by joseluis