joseluis/huion-linux-drivers #74: Pen cursor way off

zachattackrandom, 2020-05-01:
when my pen as at the right side of the side its fine but as i move it to left left an moves further and further it doesn't do this when i have my monitor unplugged i'm running manjaro with a huino kamvas 13 pro(gt133)

lululock71, 2020-05-01:
Do you have multi monitors plugged into your system?

zachattackrandom, 2020-05-02:
> Do you have multi monitors plugged into your system?

if the drawing tablet counts as one then yes i have my main monitor then my drawing tablet i have tried enabling multi monitor and nothing changes set the resolotuion etc maybe its because my main monitor 1920x1200 wich is wierd idk

lululock71, 2020-05-02:
Try forcing mirror mode on your screens. You might have top/bottom margins on your tablet, due to the difference in resolution, or just force your main monitor to display in 1920x1080p. 

I currently have the same issue, except that my main monitor has the same resolution than the tablet. I couldn't get multi monitor to work, so I got used to it and just use the tablet in mirror mode for now, waiting for a multi monitor fix.

zachattackrandom, 2020-05-02:
> Try forcing mirror mode on your screens. You might have top/bottom margins on your tablet, due to the difference in resolution, or just force your main monitor to display in 1920x1080p.
>
> I currently [...]

that works but the only reason i want my main monitor on is to have refrences is there anyway to make it work with them not being mirrored

lululock71, 2020-05-02:
Yeah, I know, it's kinda annoying, but multi-monitor don't work for now.
For now, I've set up my laptop next to me on my desk when I need a second monitor but it's annoying, yes.

I've opened an issue because it made the driver crash for me, maybe you can contribute to that one : issue https://github.com/joseluis/huion-linux-drivers/issues/75

Desulate, 2020-06-25:
If you manually use xrandar you may be able to fix these issues. 

xrander uses monitors in a way you probably wouldn't expect, the basic concept is that it creates the 'total screen' area first and then positions your monitors 'screens' over that area to determine what they are actually outputting and displaying on the monitors. 

For your use-case I recommend you overlap the tablet 'screen' output over the external monitors 'screen' output and set the digitizer to match the tablets 'screen' position. Whatever extra difference in 'visual space' there is between the two monitors will simply just be cropped off from the tablets 'screen' leaving roughly 1:1 scale in dimensions between the tablets 'screen' and external monitors 'screen'. If you use some other preset 'mirroring' setting the dimensions could be slightly out of proportion making your pens input feel off.

Using the 'mirroring' mode available in some desktop environments may mess up the dimensions between the screens because they might scale certain things differently to make the environment more consistent for the majority of users, at the cost of some almost un-noticeable precision. Which wouldn't normally be an issue but we have an extreme use-case here where were trying to precisely match 2 varying monitors and a digitizer for precision raster drawing.

For example here are my current settings: 
```
[monitor_2+1]
# 3 monitors. But 2 and 3 share the same view

screen_1W = 1920
screen_1H = 1080
screen_2W = 1920
screen_2H = 1200
screen_3W = 1920
screen_3H = 1080

total_screen_width   = ${screen_1W} + ${screen_2W}
total_screen_height  = ${screen_2H}
tablet_offset_x      = ${screen_1W}
tablet_offset_y      = 60

xrandr_output1 = --output DisplayPort-0 --mode ${screen_1W}x${screen_1H} --pos 0x0 --rotate normal
xrandr_output2 = --output DisplayPort-1 --mode ${screen_2W}x${screen_2H} --pos ${screen_1W}x0 --rotate normal
xrandr_output3 = --output HDMI-A-0 --mode ${screen_3W}x${screen_3H} --pos ${screen_1W}x60 --rotate normal
xrandr_args    = ${xrandr_output1} ${xrandr_output2} ${xrandr_output3}
```
I have 3 monitors but I centered my Huion kamvas 12 pro's monitor [HDMI-A-0] in the center of my 1920x1200 wide gamut monitor [DisplayPort-1], and since this script sets the digitizer area to match the tablets hardware I only had to position it to match the tablets screen position. 

@zachattackrandom You'd want to try doing something similar, like this for example:

```
[monitor_2in1]
# 2 monitors. But 2 is centered in 1

screen_1W = 1920
screen_1H = 1200
screen_2W = 1920
screen_2H = 1080

total_screen_width   = ${screen_1W} 
total_screen_height  = ${screen_1H}
tablet_offset_x      = 0
tablet_offset_y      = 60

xrandr_output1 = --output DisplayPort-0 --mode ${screen_1W}x${screen_1H} --pos 0x0 --rotate normal
xrandr_output3 = --output HDMI-A-0 --mode ${screen_2W}x${screen_2H} --pos 0x60 --rotate normal
xrandr_args    = ${xrandr_output1} ${xrandr_output2}
```

zachattackrandom, 2020-06-25:
Thanks for the help but I ended up just setting up a windows vm to use my drawing tablet with since it's easier and the driver is better than anything we have on Linux

closed 2020-06-25 by zachattackrandom