joseluis/huion-linux-drivers #75: Multi monitor doesn't work, throws an error

lululock71, 2020-05-02:
Hi,
I wanted to experiment (again) with multi monitor setup to finally get it to work, after the latest update of the .py file. But it doesn't work.

**My current setup :**

- ArchLinux 5.4.35-1 LTS
- Pyhton 3.8.2
- Xorg 1.20.8-1
- All dependencies installed (the driver works fine when multi monitor isn't enabled)
- Huion Kamvas Pro 13 (GT-133)
- Main monitor is 1080p, plugged in DVI-D port

**Issue symptoms :**

Error leading to script crash.
```
sudo ./huion-tablet-driver.py
Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
Setting up driver. . . Done!
	Tablet model name         GT-133
	Buttons                   ENABLED (5)
	Scrollbar                 ENABLED (1)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    Enabled (1920x1080)
	Current Monitor Setup     [monitor_2]
	Multi Monitor Setup       ENABLED
	Calling xrandr            ENABLED

Setting up multiple monitors. . . 
Running xrandr. . . 
Running xinput. . . 
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  131 (XInputExtension)
  Minor opcode of failed request:  57 ()
  Serial number of failed request:  20
  Current serial number in output stream:  21
ERROR running the following comand:
	xinput set-prop "Tablet Monitor Pen 094346" --type=float "Coordinate Transformation Matrix" 0.5 0 0.5 0 1.0 0.0 0 0 1
RETURN CODE: 1
Exception ignored in: <function InputDevice.__del__ at 0x7ff70f804040>
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/evdev/device.py", line 159, in __del__
  File "/usr/lib/python3.8/site-packages/evdev/device.py", line 304, in close
  File "/usr/lib/python3.8/site-packages/evdev/eventio_async.py", line 54, in close
  File "/usr/lib/python3.8/asyncio/events.py", line 722, in get_event_loop_policy
  File "/usr/lib/python3.8/asyncio/events.py", line 715, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down

```

**config.ini file used :**
```
[config]

# Your tablet. Find the supported models at the end of this file.
# Or use [tablet_debug] for just printing input info for sharing.
current_tablet = [tablet_gt133]
debug_mode     = false

# Configure buttons
enable_buttons          = true
pen_buttons_reverse     = false

# Configure scrollbar
enable_scrollbar        = true
scrollbar_reverse       = false

# Multi Monitor Configuration
enable_multi_monitor    = true
enable_xrandr           = true
current_monitor_setup   = [monitor_2]

# Calibration Data
enable_calibration      = false
calibrate_min_x         = 250
calibrate_max_x         = 95250
calibrate_min_y         = 300
calibrate_max_y         = 53400

# Notifications
enable_notifications    = true
buttons_notifications   = true
scrollbar_notifications = false

# Miscellaneus
uclogic_bins            = /usr/bin
refresh_rate_fps        = 300

# Here you can select a menu with the appropriate number of buttons for your tablet. E.g.:
#start_menu             = [menu_simple_4b]
#start_menu             = [menu_main_10b]
start_menu              = [menu_krita_5b]

[menu_krita_5b]
b0 = key ctrl+s       # save file
b1 = key b            # brush tool (Krita)
b2 = key ctrl+shift+z # redo
b3 = key ctrl+z       # undo
b4 = key e            # eraser toggle (Krita)
#
# scrollbar (up/down)
su = key ctrl+minus   # zoom out (krita)
sd = key ctrl+plus    # zoom in (krita)

[monitor_2]
# 2 monitors arranged horizontally

# screens widths and heights
screen_1W = 1920
screen_1H = 1080
screen_2W = 1920
screen_2H = 1080

# global geometry
total_screen_width   = ${screen_1W} + ${screen_2W}
total_screen_height  = ${screen_1H}
tablet_offset_x      = ${screen_1W}
tablet_offset_y      = 0

xrandr_output1 = --output DVI-D-0 --mode ${screen_1W}x${screen_1H} --pos 0x0 --rotate normal
xrandr_output2 = --output HDMI-A-0 --mode ${screen_2W}x${screen_2H} --pos ${screen_1W}x0 --rotate normal
xrandr_args    = ${xrandr_output1} ${xrandr_output2}

[tablet_gt133]
model_name    = GT-133
screen        = true
screen_width  = 1920
screen_height = 1080
pen_max_x     = 58752
pen_max_y     = 33048
pen_max_z     = 8191
resolution    = 5080
buttons       = 5
scrollbar     = 1
pen_min_tilt  = -60
pen_max_tilt  = 60
```

joseluis, 2020-05-02:
Hi @lululock71 I tried with python 3.8 and it doesn't work, neither with python3.7. I must investigate that, but for now the fastest solution for you and for now is to use it with python 3.6. Two easy steps:
1. `sudo apt install python-3.6'
2. and then change the first line of your copy of the script for: `#!/usr/bin/env python3.6`

joseluis, 2020-05-02:
I've been looking and this is where I got:
* I can't import the evdev module anymore in python3.7 or greater. My error is different than yours and I don't know if it's all system dependent, but this kind of things shouldn't happen, nor the bugs people have in their different OSes due to the unreliability of the python ecosystem configuration...
* Something [on this list of changes since 3.6](https://docs.python.org/3/whatsnew/3.7.html#porting-to-python-37) may have something to do but I can't see it, and I really don't want to spend any time diving into this rabbit hole.

This actualy gives me more confidence regarding what I was deciding lately about rewriting this whole project in Rust, refactoring to make it more reliable and maintainable... Leaving python behind would make me very happy... I think this may be the time to do it.

lululock71, 2020-05-03:
I'm on Arch, so my Python interpreter is always updated to the latest version and 3.6 is only available in AUR. I will not use multi monitor for now. It always been broken anyway...

> This actualy gives me more confidence regarding what I was deciding lately about rewriting this whole project in Rust, refactoring to make it more reliable and maintainable... 

Rust is an awesome and reliable language but why not C ? More people would be capable of helping and the development will be faster... Are you sure that the advantages brought by Rust over C will make the difference for this quite small project ? Also, C compilers are more robust and reliable nowadays.

joseluis, 2020-05-03:
> Rust is an awesome and reliable language but why not C ? 

Honestly, mainly because after learning Rust and enjoying its features and ecosystem for a while now, I longer desire to program in any other language for the next 50 years :smile: Otherwise I really can't see myself keeping motivated to maintain this neither in python, C, haskell, go or whatever. And if I use the language I'm most comfortable with, and have fun doing it, it will only lead to good things.

I also believe it's sufficiently mature right now for this task and more. Even providing it of a simple GUI. But for now I'm gonna just try to recreate the current functionality in a more sensible way.

donbowman, 2020-06-08:
> I've been looking and this is where I got:
>
> * I can't import the evdev module anymore in python3.7 or greater. My error is different than yours and I don't know if it's all system dependent, but [...]

I have no problem import evdev into python 3.8.2
```
Python 3.8.2 (default, Apr 27 2020, 15:53:34) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import evdev
```
on ubuntu 20.04

lululock71, 2020-06-10:
I think it depends on the distro you're using. It's working under Ubuntu but on Arch, the module location might be different, hence that error. It's completely system dependent.

That's why the script should be able to detect the location of the module, before trying to load it.