joseluis/huion-linux-drivers #77: H320M Debug showing pressure and location input, no cursor on display.

rosagi-png, 2020-05-08:
Hi, attempting to run H320M on an x230 laptop, receiving the following xinput error message below. 
When running huion-tablet-driver.py, debug shows positioning and pressure input report, however the cursor doesn't appear on display (monitor 2 setup values replaced with input and resolution -as reported from xrandr). Thank you for your efforts

 XINPUT: ERROR running the following comand:
	xinput list --short | grep "Tablet Monitor Pen"
RETURN CODE: 1

Log as follows --

Finding USB device. . . Done!
Reading configuration. . . Done!
Preparing driver. . . Done!
--------------------------------------------------------------------------------
    |           Manufacturer: HUION
    |                Product: Huion Tablet_H320M
 64 |        Params block #1: ???????????????????????????????????????????????????????????????????????????????????????????????????
    |                          Max X: 8192
    |                          Max Y: 357
    |                   Max pressure: 2301
    |                     Resolution: 2048
    |
 79 |         Internal model: ???????????????????????????????????????????????????????????????????????????????????????????????????
 7b |         Buttons status: ???????????????????????????????????????????????????????????????????????????????????????????????????
 c8 |        Params block #2: ??o?????
    |                          Max X: 45720
    |                          Max Y: 28580
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_T198_190924
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
--------------------------------------------------------------------------------
Setting up driver. . . Done!
	Tablet model name         H320M
	Buttons                   ENABLED (11)
	Scrollbar                 disabled (0)
	Notifications:            ENABLED
		for buttons       ENABLED
		for scrollbar     disabled
	Screen                    disabled

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
			< DEBUG MODE ENABLED >
Enabled by default. You can disable it by setting debug_mode = false
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

SYSTEM: Linux 4.15.0-54-generic (x86_64 x86_64)
#58-Ubuntu SMP Mon Jun 24 10:55:24 UTC 2019

DEVICE: /dev/input/event19 (py-evdev-uinput)
bus: 0003, vendor 0001, product 0001, version 0003

      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

TABLET CAPABILITIES:
[('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)]
[(['BTN_DIGI', 'BTN_TOOL_PEN'], 320), ('BTN_TOUCH', 330), ('BTN_STYLUS', 331), ('BTN_STYLUS2', 332)]
[(('ABS_X', 0), AbsInfo(value=0, min=0, max=44200, fuzz=0, flat=0, resolution=5080)), (('ABS_Y', 1), AbsInfo(value=0, min=0, max=27600, fuzz=0, flat=0, resolution=5080)), (('ABS_PRESSURE', 24), AbsInfo(value=0, min=0, max=8191, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_X', 26), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0)), (('ABS_TILT_Y', 27), AbsInfo(value=0, min=0, max=0, fuzz=0, flat=0, resolution=0))]

VPEN:
name "Tablet Monitor Pen 155127", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
ERROR running the following comand:
	xinput list --short | grep "Tablet Monitor Pen"
RETURN CODE: 1

Huion Kamvas driver should now be running. . .

(Input from the tablet will be printed out)

[menu_4b_simple]
button 0 = key Tab           # hide interface
button 1 = key r             # rect select (gimp) & pick layer (krita)
button 2 = key ctrl+z        # undo
button 3 = key ctrl+y        # redo (gimp)
button 4 = 
button 5 = 
button 6 = 
button 7 = 
button 8 = 
button 9 = 
button 10 =

rosagi-png, 2020-05-08:
Fixed! It appears the issue was resolved by removing and reinstalling xinput so it works with python 3.6 rather than 2.7, similar to two other dependency issues I came across - with evdev and numexpr, however these were resolved by installing with pip3. Thank you all the same!

joseluis, 2020-05-08:
very interesting information, it could be useful to others, thank you

closed 2020-05-08 by joseluis