joseluis/huion-linux-drivers #78: Xinput error 1060Plus. Ubuntu 20.04

climent1, 2020-05-13:
Hi! after installation I get the next error in "xinput" in sudo ./huion-tablet-driver.py

VPEN:
name "Tablet Monitor Pen 210950", bus "BUS_USB", vendor "0001", product "0001", version "0003", phys "py-evdev-uinput"
event types: EV_SYN EV_KEY EV_ABS

XINPUT:
ERROR running the following comand:
	xinput list --short | grep "Tablet Monitor Pen"

joseluis, 2020-05-14:
Unfortunately there are bugs that appeared in recent versions of linux distributions, between different versions of python that are complicated to sort out.

I'm currently recreating the software in Rust, and the bugs will all be addressed in this new version.

climent1, 2020-05-14:
thanks a lot. I will be watching the news

closed 2020-05-14 by climent1