joseluis/huion-linux-drivers #82: Support for GS-1331 (Kamvas 13)

CammiePone, 2020-06-18:
I've got the new Kamvas 13 (not the pro), and I'm really wanting to switch to Linux. Unfortunately, the drivers for my tablet and the art software are the only things I use preventing this switch.

I don't know how hard it would be to add the support, or if it would just work with one of the similar tablets in the config.ini, but I didn't see the Kamvas 13 in it, so I'm assuming it won't work.

If not, that's alright. I understand that it's probably a lot of work for a tablet not a lot of people are currently using.

obook, 2020-06-19:
Hello CammiePone,
For Kamvas 13 may you test in config.ini
`current_tablet = [tablet_gt133]`
?
Olivier, Kamvas Pro 13 user under Ubuntu and Raspberry

CammiePone, 2020-06-19:
I can try next chance I get, but I don't expect it to work perfectly. The Kamvas 13 uses a newer pen, has two buttons instead of a slider, and has some extra functionality that the Kamvas Pro 13 lacks.

obook, 2020-06-19:
Hello CammiePone,
I think is the same pen. The slider is on a border of tablet for pro version.
The important thing is compile uclogic, install all requested softwares (xinput xserver-xorg-input-evdev python3-evdev python3-usb xdotool libnotify-bin arandr python3-numexpr) then create the xorg config file.
Good luck,
Olivier

CammiePone, 2020-06-19:
The Kamvas Pro 13 uses the PW507,  whilst the Kamvas 13 uses the PW517. The touch bar of the pro was replaced by two extra buttons on the regular Kamvas 13, meaning I'd likely lose out on functionality there. While I'm sure I could get a hackjob working, I'd be missing pieces that I consider vital to drawing, hence why I made the issue to add support for the tablet.

yoanndaly, 2020-06-30:
I've also got the new Kamvas 13 and I use Ubuntu 18.04. I only suggests a slight improvement to the solution suggested by Olivier.
I wrote the minimal variant of the configuration for Kamvas Pro 13 with corrected button information which goes as follows:

`[tablet_gs1331]
model_name    = GS1331
screen        = true
screen_width  = 1920
screen_height = 1080
pen_max_x     = 58752
pen_max_y     = 33048
pen_max_z     = 8191
resolution    = 5080
buttons       = 8
pen_min_tilt  = -60
pen_max_tilt  = 60`

When I execute the script, the buttons work quite well (of course, after defining a new menu with 8 buttons) but the pen is not recognized before restart. In debug mode though, the printing instructions for the positions of the pen appear in the terminal but the mouse pointer is not moved.  After reboot, the pen PW517 works quite well, it seems to be calibrated even in the corners, but the buttons aren't working anymore. If I run the script again after reboot, it disables the pen and the 8 buttons work again.
Any idea on how to fix this issue would be welcome.
Best,
Yoann

yoanndaly, 2020-07-01:
I continue  my previous comment. I have finally solved my installation issue of my Kamvas 13, as explained in a comment of the closest bug I have found [#41 ](https://github.com/joseluis/huion-linux-drivers/issues/41). So I can confirm that if one follows the installation instructions for the driver, with the above configuration I suggested yesterday `current_tablet = [tablet_gs1331]`, then both the buttons and pen work fine and are properly calibrated (at least in miror display). Note that, with respect to the Kamvas Pro 13, I removed the line `scrollbar     = 1` since there is no scrollbar on the Kamvas 13. Note also that the calibration values are exactly those given by the debug mode (based on uclogic-decode), so that they should really be the correct values.

Good luck,
Yoann

CammiePone, 2020-07-01:
Wait wait, so you got it perfectly function, Yoann? Nothing's missing?

yoanndaly, 2020-07-02:
Yes, I think I got it perfectly function. The script works as it is for me. I didn't try to use the monitor settings of the script in config.ini yet. I am used to setting several screens by hand. This was easier at first, but this will definitely help to include my preferred settings once and for all there and I am confident this will work fine too.